HipsterBrown/xs-dev #1: a few notes

phoddie, 2022-01-05:
I spent a little time trying this out. It is very promising.  I don't know much about Node and even less about zx,  but the script is still pretty straightforward to follow. That's great.

The following notes are based on a naive first impression:

- If $MODDABLE is already defined, nothing much works. I'm not sure how to solve that, but perhaps a warning at least?
- Is there a way to unsetup? Since the .profile is modified automatically, it could be nice to store it with the tool somehow.
- The `clone` command in the readme.md does a `cd` into `xs-dev` instead of `xs-setup` (`git clone https://github.com/HipsterBrown/xs-setup && cd xs-dev`)
- On an x86 Mac, the ESP8266 and ESP32 installs fail with:
   ```
   $ arch -arm64 brew install python; arch -arm64 brew upgrade python
   arch: Unknown architecture: arm64
   Error: arch: Unknown architecture: arm64
       at file:///Users/hoddie/Projects/xs-setup/xs-dev.mjs:174:15
       exit code: 1
   ```
- `./xs-dev.mjs update` does a full rebuild even if nothing changed
- ` ./xs-dev.mjs test` works but ` ./xs-dev.mjs run-example helloworld` exits after printing:
   ```
    ./xs-dev.mjs run-example helloworld
   ```

 I'm assuming some of this is operator error, so I put it all in one issue. If you prefer, I can break it out into separate issues.

HipsterBrown, 2022-01-05:
Thanks for trying this out and providing feedback! These are great notes for making this tool less naive and more production-ready for Moddable devs. 

1. That's a good point about systems with the `MODDABLE` repo already and should be a quick fix by looking for that env variable before defaulting to `~/.local/share/moddable`. 

2. This would be interesting to address because I'm not sure how much to "unsetup." Would it just remove the `moddable` repo and dependent env variables? Should it support reversing all the platform tooling setup too?

3. That's definitely a typo on my part after rethinking the name of this to be `xs-dev` 😅 

4. Another assumption on my part as I worked on debugging local issues on my ARM Mac, should be fine to remove this `arch` prefix

5. This is definitely naive command that could be optimized to look for remote changes before pulling and rebuilding. 

6. I should have noted in the readme that command is still a work in progress 😅 (https://github.com/HipsterBrown/xs-setup/blob/main/xs-dev.mjs#L117)

---

If you have any other workflow feedback from clients or other folks in the community, I'd like to build that in to the roadmap for this tool or help contribute it into the core moddable tooling if it makes sense. Thanks again for collaborating!

phoddie, 2022-01-05:
Thanks for the quick reply.

2. For "unsetup" I was mostly thinking about removing the added environment variables. Removing the repositories could be destructive (changes made to Moddable SDK examples), so maybe a good compromise is to display where they are so a developer can easily remove them if desired.
4. Removing the `arch` prefix works. Both esp8266 and esp32 build, install, run, and debug for me now!
7. There doesn't seem to be a way to specify the `-p` argument. To be able to run the Piu examples, for example, the platform needs have a screen (such as `esp/moddable_one` or `esp32/moddable_two`).
8. It looks like `xs-dev` always sets the `UPLOAD_PORT`. The value it sets wrong on my system (of course!) and the Moddable SDK default would have worked. Maybe `xs-dev` shouldn't have a default, but provide a way for the user to configure one if needed? (FWIW – `esptool.py` can scan all serial ports looking for an ESP8266 or ESP32, which could be a useful tool to find the port a device is attached to)

HipsterBrown, 2022-01-06:
I addressed most of these notes in a single commit (https://github.com/HipsterBrown/xs-setup/commit/0354fd37fa8ecb0feb70508427a57e23674a2293). 

I'd like to break out the "unsetup" discussion into a separate issue for consideration and planning. Otherwise, the next major milestone I'd like to work on is a refactor into a "real CLI" that can be installed and called globally while DRY-ing out some of the common functionality. It should make it more accessible for contributors to add support for other platforms, like Windows or Linux, and devices, like the Pi Pico or Gecko boards.

closed 2022-01-06 by HipsterBrown

phoddie, 2022-01-06:
Thanks! That is working better. A couple minor details:

- There's a `uest` instead of `test` in the "Moddable SDK successfully set up!" message
- One `arch -arm64` remains in step 6 ("ensure python, pip, and pyserial are installed")