HipsterBrown/xs-dev #2: esp tools directory

phoddie, 2022-01-06:
xs-setup installs the ESP32 tools to `~/.local/share/` but the ESP8266 tools are  at `~/esp`. Using the [ESP_BASE](https://github.com/Moddable-OpenSource/moddable/blob/83dadd3def6d2e7e75fc003a5ab409aa81275dd8/tools/mcconfig/make.esp.mk#L20) environment variable it looks possible to move the ESP tools to `~/.local/share/` as well. 

The `ESP_BASE` name is a little too generic as it was added when there was only ESP8266. It isn't much used. What do you think about renaming it to `ESP8266_BASE` or even `MODDABLE_ESP8266_BASE`?

HipsterBrown, 2022-01-06:
I originally tried to put the `esp` directory in `~/.local/share` but then saw the `~/esp` path hard coded in the Moddable tooling. I didn't know about the `ESP_BASE` env, so that's a nice fix. 

If there is a renaming, it would be nice to keep it consistent among all the env vars, I.e. using `_PATH` or `_BASE` as the conventional suffix.

HipsterBrown, 2022-01-06:
The use of `esp` as the platform identifier for the esp8266 is one of the reasons I added the alias for the "device" flag.

phoddie, 2022-01-06:
`_PATH` seems a little more common and clear, so `MODDABLE_ESP8266_PATH` would be the most explicit and clear.

For consistency, we could try putting the aliasing of `esp8266` to `esp` in `mcconfig` and `mcrun`. There's [already something like](https://github.com/Moddable-OpenSource/moddable/blob/83dadd3def6d2e7e75fc003a5ab409aa81275dd8/tools/mcmanifest.js#L1283-L1298) that for `sim`/`simulator`. That would work for the command line but inside manifests, platform specific sections would still use `esp`.

HipsterBrown, 2022-01-06:
> platform specific sections would still use esp

Is this because of how platform-specific dependencies are pulled in by mcconfig?

phoddie, 2022-01-06:
`mcconfig` doesn't really pull in the platform dependencies. That is controlled by the manifest. The platform section of a manifest matches against the platform. Here's [a fragment](https://github.com/Moddable-OpenSource/moddable/blob/83dadd3def6d2e7e75fc003a5ab409aa81275dd8/examples/commodetto/epaper-mini-travel-time/manifest.json#L36-L41):

```json
	"platforms": {
		"esp/moddable_three": {
			"config": {
				"rotation": 90
			}
		},
```

The resolution of manifests chains through those platform sections. The base manifest [pulls in the esp platform here](https://github.com/Moddable-OpenSource/moddable/blob/83dadd3def6d2e7e75fc003a5ab409aa81275dd8/examples/manifest_base.json#L40-L42):

```json
	"platforms": {
		"esp": {
			"include": "$(BUILD)/devices/esp/manifest.json",
```
The esp platform manifest uses the platform section to [pull in the subplatform manifest](https://github.com/Moddable-OpenSource/moddable/blob/83dadd3def6d2e7e75fc003a5ab409aa81275dd8/build/devices/esp/manifest.json#L31-L34).

```json
	"platforms": {
		"esp/*": {
			"include": "./targets/$(SUBPLATFORM)/manifest.json"
		}
```

To make all those use esp8266 would be much more complicated,  and not that big a benefit. A top level alias like you did seems like a reasonable start. That's a simple change in `mcmanifest.js`:

```
				if ("esp8266" === parts[0])
					parts[0] = "esp";
				else if ((parts[0] == "sim") || (parts[0] == "simulator")) {
```

HipsterBrown, 2022-01-09:
This feature has been implemented in https://github.com/HipsterBrown/xs-setup/commit/facd1af3806b5ccde1f117eaae36f5b6eef033c3, so all cloned / curled / installed dependencies are collected under `~/.local/share`. 

Updating the env variables with a consistent `_PATH` suffix and aliasing `esp8266` references to `esp` will be moved to a discussion in https://github.com/Moddable-OpenSource/moddable

closed 2022-01-09 by HipsterBrown

phoddie, 2022-01-09:
The ESP8266 install now goes into `~/.local/share`. That's great.

One small detail: my first run of `./xs-dev.mjs setup --device=esp8266` failed because I didn't have `wget` installed. I'm really not a web developer. ;)

HipsterBrown, 2022-01-10:
Ah good note. I'll add that to the list of dependencies for that action. Thanks for testing! 😄