HipsterBrown/xs-dev #3: feat: create "mod-export.sh" for easier cleanup of dev environment

HipsterBrown, 2022-01-09:
As mentioned in #1, @phoddie mentioned:

> Is there a way to unsetup? Since the .profile is modified automatically, it could be nice to store it with the tool somehow.

Since this CLI tool appends several exports and modifies the global PATH, cleaning up those changes could be a hassle to automate. It is also a "leaky abstraction" to have all of those changes visible to the end user, who might be familiar with maintaining their shell profile. 

If those changes were collected under a single `mod-export.sh` (similar to the `export.sh` file used by the `esp-idf` tooling), then appending a single `source $MOD_PATH/mod-export.sh` (`MOD_PATH` being somewhere under the `~/.local/share` directory with the other installed dependencies) to the shell profile will be much easier to maintain long term and remove through a potential `xs-dev teardown` command.

HipsterBrown, 2022-01-18:
Resolved with the release of version 0.1.0: https://github.com/HipsterBrown/xs-dev/blob/main/CHANGELOG.md#010

closed 2022-01-18 by HipsterBrown

phoddie, 2022-01-19:
This is working nicely and looks like a clean solution. I did run into a problem trying it out. My Mac didn't have a `~/.bashrc` file led to an error. I worked around it but I'm not sure I got it right.

HipsterBrown, 2022-01-19:
@phoddie can you share the error output? Where do you normally put your shell environment config?

I can update the shell profile logic to ensure the assumed file exists.

phoddie, 2022-01-19:
Here's the output:

```
jphAir2018:promiscuous hoddie$ xs-dev setup
Setting up the mac tools!
✔ Cloning Moddable-OpenSource/moddable repo
/usr/local/lib/node_modules/xs-dev/node_modules/gluegun/build/index.js:13
    throw up;
    ^

Error: file not found /Users/hoddie/.bashrc
    at /usr/local/lib/node_modules/xs-dev/node_modules/gluegun/build/toolbox/patching-tools.js:183:23
    at step (/usr/local/lib/node_modules/xs-dev/node_modules/gluegun/build/toolbox/patching-tools.js:33:23)
    at Object.next (/usr/local/lib/node_modules/xs-dev/node_modules/gluegun/build/toolbox/patching-tools.js:14:53)
    at /usr/local/lib/node_modules/xs-dev/node_modules/gluegun/build/toolbox/patching-tools.js:8:71
    at new Promise (<anonymous>)
    at __awaiter (/usr/local/lib/node_modules/xs-dev/node_modules/gluegun/build/toolbox/patching-tools.js:4:12)
    at readFile (/usr/local/lib/node_modules/xs-dev/node_modules/gluegun/build/toolbox/patching-tools.js:179:12)
    at Object.<anonymous> (/usr/local/lib/node_modules/xs-dev/node_modules/gluegun/build/toolbox/patching-tools.js:86:46)
    at step (/usr/local/lib/node_modules/xs-dev/node_modules/gluegun/build/toolbox/patching-tools.js:33:23)
    at Object.next (/usr/local/lib/node_modules/xs-dev/node_modules/gluegun/build/toolbox/patching-tools.js:14:53)
```

> Where do you normally put your shell environment config?

`~/.profile`