HipsterBrown/xs-dev #12: feat: "scan" command or flag

HipsterBrown, 2022-02-03:
While the underlying tooling for ESP32 and ESP8266 generally works to discover dev board ports, it would be helpful to provide a command or flag to scan from xs-dev and display a nice output. 

This could start by shelling to esptool.py and parsing the output. It would be even better to find a way of scanning from NodeJS; perhaps through serialport, node-usb, or something else like the espressif.js utility. 

Once pico support lands in the main branch, this command could expand to look for those boards as well.

A nice to have after scanning would be to save a discovered port to some project configuration file, something like the `.env` file described in another issue.

HipsterBrown, 2022-02-05:
As I continue to work with esptool, I realized there might be a path toward auto-detecting the device type from the `chip_id` logic in that tool. This could simplify the need for the repetitive `--device` flag when running a project. 🤔 
If this works, the default behavior for `xs-dev run` would be to detect the device type instead of running the platform simulator. That behavior could be replaced by a `--sim` flag, rather than continuing to consider the `mac`, `win`, `lin`, and `wasm` targets as "devices". 

One potential issue to consider is the ability to detect / select device subsets, i.e. `esp32/moddable_two`. 🤷

phoddie, 2022-02-10:
> So commands could update to xs-dev run --sim wasm instead of xs-dev run --device wasm. Or default to the OS platform without an argument: xs-dev run --sim. I think xs-dev run should still default to the OS platform simulator if no --device flag is passed.

That sounds pretty reasonable, and nicely doesn't veer far from the behaviors of `mcconfig` and `mcrun`.

PR #22 (2022-02-21): feat: add scan command
Fixes #12 

- [X] Includes documentation for this new feature
- [X] Tested locally

This feature uses [SerialPort](https://serialport.io/) to list all ports and call `esptool.py` for each relevant port since `esptool.py` stops at the first available device, which is an issue if you have multiple devices connected for testing.
closed 2022-02-21 by HipsterBrown