HipsterBrown/xs-dev #23: feat: Windows support

HipsterBrown, 2022-03-21:
Making an explicit issue for this so people viewing the project know its on the roadmap. I don't have a development environment for running Windows, either natively or in a VM, so community help is required.

Open questions:

- [ ] is there a popular package manager upon which xs-dev can depend, i.e. `brew` for Macos or `apt` for Debian linux? Some available options from basic research: [chocolately](https://chocolatey.org/), [scoop](https://scoop.sh/), [winget](https://github.com/microsoft/winget-cli)
- [ ] how to manage setting up environment variables automatically, given there is no shell profile like unix systems?
- [ ] is it possible to automate the espressif tooling without using the [esp-idf windows installer](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/devices/esp32.md#win-instructions)? Or does `xs-dev` just provide some help downloading the installer and checking for those dependencies before continuing?
- [ ] how does `xs-dev` treat [WSL](https://docs.microsoft.com/en-us/windows/wsl/about)?

andycarle, 2022-03-25:
I've made some good progress on the environment variables issue.

It turns out that you can reliably shell out to [`setx`](https://docs.microsoft.com/en-us/windows-server/administration/windows-commands/setx) from Node, as long as you're only modifying the user's environment variables (not the system ones). This may require a UAC elevation (similar to `sudo` on Linux, e.g.), but I don't think there's any avoiding that over the course of the entire install.

This block of code is working for me in `setup/windows.ts`:

```ts
function setEnv(name: string, value: string): void {
  print.info(`Setting environment variable ${name}:${value}`);
  const result = spawnSync(`setx`, [name, value]);
  const stdOut = result.stdout.toString();
  const stdErr = result.stderr.toString();
  if (stdOut !== "")
    print.info(stdOut);
  if (stdErr !== "")
    print.error(stdErr);
}
```

I should have a draft Windows setup extension ready for folks to try out early next week.

andycarle, 2022-03-25:
I somewhat immediately have run into an issue with using `setx` on Windows. I'll spare this thread the details, but it isn't a great solution for setting the `PATH`. I'm looking into other options there, including banging on the registry directly.

HipsterBrown, 2022-03-26:
@andycarle I spent some time today attempting to add Windows support to the moddable-prebuild project. I achieved a successful build in the [latest release](https://github.com/HipsterBrown/moddable-prebuild/releases/tag/latest), although I don't currently have a way of verifying the output. 

The `windows-latest` environment comes [pre-configured with build tools](https://github.com/actions/virtual-environments/blob/main/images/win/Windows2022-Readme.md#visual-studio-enterprise-2022), however the most useful lesson came from the [msvc-cmd-dev action](https://github.com/ilammy/msvc-dev-cmd) for setting up the command environment for C/C++ compilation (without needing to open up the special "x86 Native Tools Command Prompt for VS" env). The source of that action might be helpful for figuring out workarounds for the `setx` PATH issues you mentioned.

andycarle, 2022-04-02:
I made good progress on Windows `xs-dev setup` today. I'll put in a draft pull request of that tonight.

> how to manage setting up environment variables automatically, given there is no shell profile like unix systems?

I have a good solution for this now. It does require an extra NPM package: [regedit](https://www.npmjs.com/package/regedit). But the functionality I built on top of that works very nicely.

> how does xs-dev treat [WSL](https://docs.microsoft.com/en-us/windows/wsl/about)?

The more I think about this, the more I'm inclined to suggest that you consider WSL as outside the scope of xs-dev. The variations between WSL1 and WSL2 are serious and breaking for a lot of the workflows we'd want to support here. And, frankly, if you're building on Windows you should usually be using the native solution instead of routing it through a VM anyways. 

I could probably be convinced otherwise, but that's where I'm leaning.

andycarle, 2022-04-02:
I've created the Initial draft PR for Windows setup support. This takes care of setting up the environment variables (including path), cloning the repo, and building tools. 

I think we're going to need platform-specific defaults in `constants.ts` to support Windows properly. Something like `C:\moddable` is going to make more sense for most users than trying to do a `.local/share` solution like you would on macOS. 

This first pass does not attempt to set up the VS tools environment for the user. It needs to be run in the VS Command Prompt (and checks for such).

andycarle, 2022-07-26:
I've put together some rough notes on how to get development versions of `xs-dev` working on Windows using `pnpm`. In case they're of use to anyone else, they're here: [https://gist.github.com/andycarle/20e0d7b331008dfd147e38f1501f17dd](https://gist.github.com/andycarle/20e0d7b331008dfd147e38f1501f17dd)

closed 2022-08-29 by HipsterBrown