HipsterBrown/xs-dev #25: Pico build fails

phoddie, 2022-03-23:
I'm trying out the Pico support on macOS. The setup succeeds without an error: `xs-dev setup --device=pico`. Running helloworld compiles the C binaries, builds the XS byte code, and links. All good.

Then it fails here:

```console
make: /Users/hoddie/pico/pico-sdk/build/elf2uf2/elf2uf2: No such file or directory
make: *** [/Users/hoddie/.local/share/moddable/build/bin/pico/debug/helloworld/xs_pico.uf2] Error 1
    at makeError (/usr/local/lib/node_modules/xs-dev/node_modules/execa/lib/error.js:60:11)
    at handlePromise (/usr/local/lib/node_modules/xs-dev/node_modules/execa/index.js:118:26)
    at processTicksAndRejections (node:internal/process/task_queues:96:5) {
  shortMessage: 'Command failed with exit code 2: mcconfig -d -m -p pico',
```

There is an older copy of the `pico-sdk` on my computer at that path, but it is not linked to by any environment variable. The copy of `pico-sdk` installed by xs-dev is at `/Users/hoddie/.local/share/pico/pico-sdk` and does appear to contain `elf2uf2` at the expected path.

phoddie, 2022-03-23:
This looks correct:

```console
$ echo $PICO_SDK_PATH
/Users/hoddie/.local/share/pico/pico-sdk
```

Perhaps the problem is in the Moddable SDK Pico build?

phoddie, 2022-03-23:
In `mcconfig/make.pico.mk`, the Pico SDK is taken from `$PICO_SDK_DIR`:

```
PICO_SDK_DIR ?= $(HOME)/pico/pico-sdk
```

xs-dev doesn't define `PICO_SDK_DIR` but `PICO_SDK_PATH`. Changing that line to this one:

```
PICO_SDK_DIR = $(PICO_SDK_PATH)
```

Then it builds, installs, runs, and debugs!

I don't think this is the correct change. I think xs-dev should use `PICO_SDK_DIR` instead of `PICO_SDK_PATH`.  

Adding @mkellner here, as he likely knows best.

mkellner, 2022-03-23:
Yes. `mcconfig/make.pico.mk` is using the environment variable `PICO_SDK_DIR`. It would probably be best for xs-dev to use `PICO_SDK_DIR`.

HipsterBrown, 2022-03-23:
This fix was released in v0.10.0 of xs-dev

PR #28 (2022-03-23): feat: teardown pico install, update to PICO_SDK_DIR
Fixes #25 
Fixes #26 

This also adds the `fontbm` install directory to the set of `teardown` actions.
closed 2022-03-23 by HipsterBrown