HipsterBrown/xs-dev #29: ^C doesn't stop build

phoddie, 2022-03-24:
On macOS, start a build with

```console
> xs-dev run --example piu/balls
```

Press Control C. The terminal prompt appears, as expected, but the build continues in the background, eventually launching xsbug and the simulator.

closed 2022-03-26 by HipsterBrown

HipsterBrown, 2022-03-28:
Yea I removed the build output for now because the process would still output a little bit after the CTRL-C signal. 

It might be helpful to create a `--debug` or `--verbosity` option for folks that want to see it.

phoddie, 2022-03-28:
Makes sense.

FWIW – a behavior I've seen to output a single line that replaces itself. Parts of the ESP-IDF build for ESP32 do this. I like it because it shows the activity without generating a huge amount of output in the terminal.

PR #35 (2022-03-26): feat: include pico in run and setup device list, add simulators list
Fixes #29 

Also resolves displaying the correct device alias for the current OS platform in the `setup` command's list of devices.