HipsterBrown/xs-dev #48: Common Core Framework

dashcraft, 2022-04-25:
Something I've been doing recently with the company I cofounded, is creating a common core application that runs our primary code. It handles initial server setup, ble wifi credential setting (improv-wifi spec), barebones/basic MQTT connection to home assistant, ota updates, etc. I wonder if it doesn't potentially make sense to create a stripped-down version to offer users to build from in xs-dev, that gives them some core features pre-programmed, an interface to interact with, and some compartments/mod options to extend the core app?

I imagine there are some common use cases (wifi onboarding was one for us) that pop up, some devs might like a framework with prebuilt features to start from.

HipsterBrown, 2022-04-27:
There are probably a few ways to go about this idea. The common core could be installed as a third-party module (see this new proposal #49) and instantiated by the project. Or it could be part of a starter template, similar to how [Moddable examples can be used to initialize new projects](https://github.com/HipsterBrown/xs-dev#start-a-project). 

The first option makes it easy to provide upgrades to consumers of this common core feature set, while the latter gets them up and running quicker through a pre-configured host with separate project `main`. 

How would you think about distributing such a thing without bundling it directly into `xs-dev`?

dashcraft, 2022-04-30:
I've narrowed it down to a handful of things i think most users _may_ want.

1. A way to "register" sensor readings and values, to sync with mqtt/home assistant out of the box
2. A simple HTTP server that registers a local domain, serves a simple visualization to send HTML page, probably served via cdn
3. A simple way of onboarding wifi credentials (esp32 i have ble implementation using improv-wifi) out of the box
4. Out of the box OTA support, UI

Numbers 1-3, i have a good working knowledge of how to solve and expose via a common host file, number 4 is upcoming on a project of mine but i haven't dug into it a ton.