HipsterBrown/xs-dev #65: Question: Allowing dependencies to set values in manifest.json

dtex, 2022-09-03:
This may not be something to solve in XS-Dev, but I wanted to put it out there. 

By default, Top Level Await no longer works in XS entry files. It's only available in imported modules (This decision was made for performance reasons). 

J5e examples assume TLA is available, and I'd like to keep the examples as simple as possible. No otherwise unnecessary async wrapper functions or Async IIFE's. It is possible to enable TLA everywhere by setting `main_async: 1` under 'defines' in manifest.json. 

Here's an example J5e manifest.json with `main_async: 1`:

```js
{
	"include": [
		"$(MODDABLE)/modules/io/manifest.json",
                "$(j5e)/lib/button/manifest.json",
                "$(j5e)/lib/led/manifest.json"
	],
	"modules": {
		"*": "./main"
	},
	"defines": {
		"main_async": 1
	}
}
```

So, is it possible for xs-dev to help with this setting? 

Maybe as a flag for init ```xs-dev init my-project io main-async```?

I'm happy to work on a PR, but wanted to run it by your first. Is this a welcome addition, and if so do you have anything preferable to an init flag?

HipsterBrown, 2022-09-03:
I do remember this being confusing when trying out j5e recently. I think making it an init flag, like the `--typescript` one, makes the most sense. Once we start to support third-party starter templates, this could be even easier (but that's a different feature 😉).

dtex, 2022-09-03:
Okay, I'll work that up and submit a PR. 

Thanks!

Also, I love the idea of third-party starter templates and will happily contribute some for J5e.

closed 2022-09-04 by HipsterBrown

HipsterBrown, 2022-09-04:
This has been resolved in version 0.19.0. Thanks for contributing @dtex 🙌 Can't wait to see more collaborating between j5e and xs-dev 😄