HipsterBrown/xs-dev #68: <xs-dev setup> fails on Raspberry Pi

ralphwetzel, 2022-09-28:
Hi!
I issued a `xs-dev setup` on my Raspberry Pi. This ended in with the following error message:

```
ℹ Moddable repo already installed
⠋ Installing simulator/usr/lib/node_modules/xs-dev/node_modules/gluegun/build/index.js:15
    throw up;
    ^

Error: Command failed with exit code 126: mcconfig -m -p x-lin /home/pi/.local/share/moddable/tools/xsbug/manifest.json
/home/pi/.local/share/moddable/build/bin/lin/release/mcconfig: line 3: /home/pi/.local/share/moddable/build/bin/lin/release/tools: cannot execute binary file: Exec format error
    at makeError (/usr/lib/node_modules/xs-dev/node_modules/execa/lib/error.js:60:11)
    at handlePromise (/usr/lib/node_modules/xs-dev/node_modules/execa/index.js:118:26)
    at processTicksAndRejections (node:internal/process/task_queues:96:5) {
  shortMessage: 'Command failed with exit code 126: mcconfig -m -p x-lin /home/pi/.local/share/moddable/tools/xsbug/manifest.json',
  command: 'mcconfig -m -p x-lin /home/pi/.local/share/moddable/tools/xsbug/manifest.json',
  escapedCommand: 'mcconfig -m -p x-lin "/home/pi/.local/share/moddable/tools/xsbug/manifest.json"',
  exitCode: 126,
  signal: undefined,
  signalDescription: undefined,
  stdout: '',
  stderr: '/home/pi/.local/share/moddable/build/bin/lin/release/mcconfig: line 3: /home/pi/.local/share/moddable/build/bin/lin/release/tools: cannot execute binary file: Exec format error',
  failed: true,
  timedOut: false,
  isCanceled: false,
  killed: false
}
```

Walking manually through the [described setup procedure](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/Moddable%20SDK%20-%20Getting%20Started.md#installing-2) does not create an error.

HipsterBrown, 2022-09-28:
Thanks for creating an issue! This could be an issue with the release build artifacts (https://github.com/Moddable-OpenSource/moddable/releases/tag/OS220901) given the error `cannot execute binary file: Exec format error`. 

If you try running `xs-dev setup --target-branch public`, does this work as expected? 

Also, did you already have the Moddable tooling installed locally before running `xs-dev setup`?

ralphwetzel, 2022-12-28:
@HipsterBrown -- sorry for my pending reply!
I fear that I cannot provide any further insight after the quite long span of time passed.
So it's kind of a cold case; I'm fine with closing this - knowing that there could be sth sleeping...

closed 2022-12-29 by HipsterBrown