HipsterBrown/xs-dev #69: teardown does not remove xsbug symlink (macOS)

phoddie, 2022-11-21:
After a successfully `xs-dev setup`, executing...

```
xs-dev teardown
```

...leaves the xsbug symlink the Applications folder. When reinstalling, that results in a warnings:

```
ℹ xsbug.app symlink already exists
```

phoddie, 2022-12-02:
Thanks for looking into this. This fix didn't work for me. I tracked it down to this line in teardown.js:

```js
if ((0, os_1.type)() === 'darwin') {
```

Running macOS 12.6 , the value of `(0, os_1.type)()` is `"Darwin"` not `"darwin"`. 

I'm also curious why `(0, os_1.type)()` isn't just `os_1.type()`.

HipsterBrown, 2022-12-02:
Ah I forgot about the capitalization from `os.type()`, which I usually normalize with `.toLowerCase()`. That's what happens when I add some logic _after_ manually testing 😞

PR #80 (2022-12-01): fix(teardown): remove xsbug symlink on macos
Closes #69
PR #84 (2022-12-03): fix(teardown): use correct reference for Darwin
Fixes #69
closed 2022-12-03 by HipsterBrown