HipsterBrown/xs-dev #76: ESP8266 tools download failure on macOS

phoddie, 2022-11-22:
From testing on a (nearly) fresh macOS install (Monterey, 12.3.1), the esp8266 device setup fails when downloading the tools from moddable.com. FWIW – moddable.com uses LetsEncrypt for TLS certificates which can lead to unusual certificate configurations. The tool [download URL](https://www.moddable.com/private/esp8266.toolchain.darwin.tgz) does work in Safari on the same computer, so it seems valid.

<details>

```
xs-dev setup --device=esp8266
ℹ Ensuring esp directory
⠧ Downloading xtensa toolchain/usr/local/lib/node_modules/xs-dev/node_modules/gluegun/build/index.js:15
    throw up;
    ^

<ref *1> Error: write EPROTO 8045F00901000000:error:0A000172:SSL routines:tls12_check_peer_sigalg:wrong signature type:../deps/openssl/openssl/ssl/t1_lib.c:1572:

    at WriteWrap.onWriteComplete [as oncomplete] (node:internal/stream_base_commons:94:16) {
  errno: -100,
  code: 'EPROTO',
  syscall: 'write',
  config: {
    transitional: {
      silentJSONParsing: true,
      forcedJSONParsing: true,
      clarifyTimeoutError: false
    },
    adapter: [Function: httpAdapter],
    transformRequest: [ [Function: transformRequest] ],
    transformResponse: [ [Function: transformResponse] ],
    timeout: 0,
    xsrfCookieName: 'XSRF-TOKEN',
    xsrfHeaderName: 'X-XSRF-TOKEN',
    maxContentLength: -1,
    maxBodyLength: -1,
    validateStatus: [Function: validateStatus],
    headers: {
      Accept: 'application/json, text/plain, */*',
      'User-Agent': 'axios/0.24.0'
    },
    responseType: 'stream',
    method: 'get',
    url: 'https://www.moddable.com/private/esp8266.toolchain.darwin.tgz',
    data: undefined
  },
  request: <ref *4> Writable {
    _writableState: WritableState {
      objectMode: false,
      highWaterMark: 16384,
      finalCalled: false,
      needDrain: false,
      ending: false,
      ended: false,
      finished: false,
      destroyed: false,
      decodeStrings: true,
      defaultEncoding: 'utf8',
      length: 0,
      writing: false,
      corked: 0,
      sync: true,
      bufferProcessing: false,
      onwrite: [Function: bound onwrite],
      writecb: null,
      writelen: 0,
      afterWriteTickInfo: null,
      buffered: [],
      bufferedIndex: 0,
      allBuffers: true,
      allNoop: true,
      pendingcb: 0,
      constructed: true,
      prefinished: false,
      errorEmitted: false,
[... 458 lines cut ...]
      host: 'www.moddable.com',
      protocol: 'https:',
      _redirectable: [Circular *4],
      [Symbol(kCapture)]: false,
      [Symbol(kBytesWritten)]: 0,
      [Symbol(kEndCalled)]: true,
      [Symbol(kNeedDrain)]: false,
      [Symbol(corked)]: 0,
      [Symbol(kOutHeaders)]: [Object: null prototype] {
        accept: [ 'Accept', 'application/json, text/plain, */*' ],
        'user-agent': [ 'User-Agent', 'axios/0.24.0' ],
        host: [ 'Host', 'www.moddable.com' ]
      },
      [Symbol(kUniqueHeaders)]: null
    },
    _currentUrl: 'https://www.moddable.com/private/esp8266.toolchain.darwin.tgz',
    [Symbol(kCapture)]: false
  },
  response: undefined,
  isAxiosError: true,
  toJSON: [Function: toJSON]
}

Node.js v18.12.1
```

HipsterBrown, 2022-12-22:
It appears to be the main difference between the download from the moddable.com host and all the others that xs-dev fetches is the cert uses TLS 1.2, while the others use TLS 1.3. I know Node 18+ is built with OpenSSL 3, so there could be different expectations from the default https agent used by those requests in xs-dev. 

Still exploring a fix, if possible.

phoddie, 2022-12-22:
Yikes. I hope LetsEncrypt and Node.js aren't incompatible.

HipsterBrown, 2022-12-22:
I use Lets Encrypt on my website, and my cert uses TLS 1.3. 🤷‍♂️

phoddie, 2022-12-22:
@mkellner – anything we can do on moddable.com to help with this?

HipsterBrown, 2022-12-23:
I got a little more info when trying to request this download with [Deno](https://deno.land), to see if it was just a Node issue. 

My test script:

```ts
const url = `https://www.moddable.com/private/esp8266.toolchain.darwin.tgz`

try {
  const result = await fetch(url)
} catch (error) {
  console.log('Unable to complete request')
  console.error(error)
}
```

The result:
```
TypeError: error sending request for url (https://www.moddable.com/private/esp8266.toolchain.darwin.tgz): error trying to connect: peer misbehaved: received unadvertised sig scheme RSA_PKCS1_SHA1
```

When digging into the Deno issue, [it appears to be incompatible with non-SHA256 signatures in TLS certs](https://github.com/denoland/deno/issues/12457). I can get around this error by using the `--unsafely-ignore-certificate-errors` in Deno. 

If I run this same script with Node 18+ and using the `--trace-tls` flag, I get a much more verbose output which also points to `Signature Algorithm: rsa_pkcs1_sha1 (0x0201)`. 

There is some info about renewing LetsEncrypt certs with `ecdsa` using `certbot` here: https://eff-certbot.readthedocs.io/en/stable/using.html#using-ecdsa-keys

phoddie, 2022-12-23:
I feel like we should solve this on the moddable.com side. That's outside my skills, but maybe @mkellner will have an idea.

mkellner, 2022-12-23:
Please use this link to fetch the 8266 toolchain:
https://github.com/Moddable-OpenSource/tools/releases/download/v1.0.0/esp8266.toolchain.darwin.tgz

HipsterBrown, 2022-12-23:
@mkellner  💯 That appears to work! At least there was no error during the download. I still need to try running "hello world" on a device. 

Will that release be updated with the `linux` and `windows` toolchains as well?

mkellner, 2022-12-23:
I've updated the release with `linux` and `windows` toolchains. I've also added the `cygwin` support package for windows.

closed 2022-12-23 by HipsterBrown

HipsterBrown, 2022-12-23:
This fix has been released in v0.20.0 of the CLI

PR #94 (2022-12-23): fix: update source for esp8266 toolchain
Fixes #76 

Confirmed this works on Linux and Mac. I also updated the `axios` dependency to the stable v1+ and pinned the version of node used in this project to 18.12.1 (current LTS). The CLI will continue to support maintenance & LTS versions of NodeJS until they are removed from [support by the core project](https://github.com/nodejs/release#release-schedule).