HipsterBrown/xs-dev #78: Pico build on macOS fails because pioasm missing

phoddie, 2022-11-22:
After install Pico support on macOS Monterey (12.3.1) building helloworld...

```
xs-dev run --example helloworld --device pico
```

...fails because `pioasm` can't be found.

<details>

```
$ xs-dev run --example helloworld --device pico
⠧ Building and deploying project /Users/peterhoddie/.local/share/moddable/examples/helloworld on pico/usr/local/lib/node_modules/xs-dev/node_modules/gluegun/build/index.js:15
    throw up;
    ^

Error: Command failed with exit code 2: mcconfig -m -p pico -t all -o /Users/peterhoddie/.local/share/moddable/build -d
make: /Users/peterhoddie/pico/pico-sdk/build/pioasm/pioasm: No such file or directory
make: *** [/Users/peterhoddie/.local/share/moddable/build/tmp/pico/debug/helloworld/ws2812.pio.h] Error 1
make: *** Waiting for unfinished jobs....
/Users/peterhoddie/.local/share/moddable/xs/platforms/pico/xsPlatform.c: In function 'fxCreateMachinePlatform':
/Users/peterhoddie/.local/share/moddable/xs/platforms/pico/xsPlatform.c:79:2: warning: implicit declaration of function 'modMachineTaskInit' [-Wimplicit-function-declaration]
   79 |  modMachineTaskInit(the);
      |  ^~~~~~~~~~~~~~~~~~
/Users/peterhoddie/.local/share/moddable/xs/platforms/mc/xsHosts.c: In function 'modRunMachineSetup':
/Users/peterhoddie/.local/share/moddable/xs/platforms/mc/xsHosts.c:559:2: warning: implicit declaration of function 'modInstrumentationSetup'; did you mean 'modInstrumentationSet_'? [-Wimplicit-function-declaration]
  559 |  modInstrumentationSetup(the);
      |  ^~~~~~~~~~~~~~~~~~~~~~~
      |  modInstrumentationSet_
/Users/peterhoddie/.local/share/moddable/xs/platforms/pico/xsPlatform.c: In function 'fxDeleteMachinePlatform':
/Users/peterhoddie/.local/share/moddable/xs/platforms/pico/xsPlatform.c:103:2: warning: implicit declaration of function 'modMachineTaskUninit' [-Wimplicit-function-declaration]
  103 |  modMachineTaskUninit(the);
      |  ^~~~~~~~~~~~~~~~~~~~
/Users/peterhoddie/.local/share/moddable/xs/platforms/pico/xsPlatform.c: In function 'fxReceiveLoop':
/Users/peterhoddie/.local/share/moddable/xs/platforms/pico/xsPlatform.c:462:6: warning: implicit declaration of function 'modMessagePostToMachine' [-Wimplicit-function-declaration]
  462 |      modMessagePostToMachine(current, NULL, 0xffff, doDebugCommand, current);
      |      ^~~~~~~~~~~~~~~~~~~~~~~
/Users/peterhoddie/.local/share/moddable/xs/platforms/pico/xsPlatform.c: In function 'fxInNetworkDebugLoop':
/Users/peterhoddie/.local/share/moddable/xs/platforms/pico/xsPlatform.c:723:66: warning: comparison between pointer and integer
  723 |  return the->DEBUG_LOOP && the->connection && (kSerialConnection != the->connection);
      |                                                                  ^~
/var/folders/nd/2xq97b3j4fgd9zqrz1hlykvm0000gn/T//ccxgp5iL.s: Assembler messages:
/var/folders/nd/2xq97b3j4fgd9zqrz1hlykvm0000gn/T//ccxgp5iL.s:457: Warning: setting incorrect section attributes for .data.xsram
/var/folders/nd/2xq97b3j4fgd9zqrz1hlykvm0000gn/T//ccC4rD4c.s: Assembler messages:
/var/folders/nd/2xq97b3j4fgd9zqrz1hlykvm0000gn/T//ccC4rD4c.s:2846: Warning: setting incorrect section attributes for .rodata
# compile pio:  ws2812.pio
# library xs: xsHosts.c
# library xs: xsHost.c
# library xs: xsPlatform.c
# library xs: xsAll.c
    at makeError (/usr/local/lib/node_modules/xs-dev/node_modules/execa/lib/error.js:60:11)
    at handlePromise (/usr/local/lib/node_modules/xs-dev/node_modules/execa/index.js:118:26)
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5) {
  shortMessage: 'Command failed with exit code 2: mcconfig -m -p pico -t all -o /Users/peterhoddie/.local/share/moddable/build -d',
  command: 'mcconfig -m -p pico -t all -o /Users/peterhoddie/.local/share/moddable/build -d',
  escapedCommand: 'mcconfig -m -p pico -t all -o "/Users/peterhoddie/.local/share/moddable/build" -d',
  exitCode: 2,
  signal: undefined,
  signalDescription: undefined,
  stdout: '# compile pio:  ws2812.pio\n' +
    '# library xs: xsHosts.c\n' +
    '# library xs: xsHost.c\n' +
    '# library xs: xsPlatform.c\n' +
    '# library xs: xsAll.c',
  stderr: 'make: /Users/peterhoddie/pico/pico-sdk/build/pioasm/pioasm: No such file or directory\n' +
    'make: *** [/Users/peterhoddie/.local/share/moddable/build/tmp/pico/debug/helloworld/ws2812.pio.h] Error 1\n' +
    'make: *** Waiting for unfinished jobs....\n' +
    "/Users/peterhoddie/.local/share/moddable/xs/platforms/pico/xsPlatform.c: In function 'fxCreateMachinePlatform':\n" +
    "/Users/peterhoddie/.local/share/moddable/xs/platforms/pico/xsPlatform.c:79:2: warning: implicit declaration of function 'modMachineTaskInit' [-Wimplicit-function-declaration]\n" +
    '   79 |  modMachineTaskInit(the);\n' +
    '      |  ^~~~~~~~~~~~~~~~~~\n' +
    "/Users/peterhoddie/.local/share/moddable/xs/platforms/mc/xsHosts.c: In function 'modRunMachineSetup':\n" +
    "/Users/peterhoddie/.local/share/moddable/xs/platforms/mc/xsHosts.c:559:2: warning: implicit declaration of function 'modInstrumentationSetup'; did you mean 'modInstrumentationSet_'? [-Wimplicit-function-declaration]\n" +
    '  559 |  modInstrumentationSetup(the);\n' +
    '      |  ^~~~~~~~~~~~~~~~~~~~~~~\n' +
    '      |  modInstrumentationSet_\n' +
    "/Users/peterhoddie/.local/share/moddable/xs/platforms/pico/xsPlatform.c: In function 'fxDeleteMachinePlatform':\n" +
    "/Users/peterhoddie/.local/share/moddable/xs/platforms/pico/xsPlatform.c:103:2: warning: implicit declaration of function 'modMachineTaskUninit' [-Wimplicit-function-declaration]\n" +
    '  103 |  modMachineTaskUninit(the);\n' +
    '      |  ^~~~~~~~~~~~~~~~~~~~\n' +
    "/Users/peterhoddie/.local/share/moddable/xs/platforms/pico/xsPlatform.c: In function 'fxReceiveLoop':\n" +
    "/Users/peterhoddie/.local/share/moddable/xs/platforms/pico/xsPlatform.c:462:6: warning: implicit declaration of function 'modMessagePostToMachine' [-Wimplicit-function-declaration]\n" +
    '  462 |      modMessagePostToMachine(current, NULL, 0xffff, doDebugCommand, current);\n' +
    '      |      ^~~~~~~~~~~~~~~~~~~~~~~\n' +
    "/Users/peterhoddie/.local/share/moddable/xs/platforms/pico/xsPlatform.c: In function 'fxInNetworkDebugLoop':\n" +
    '/Users/peterhoddie/.local/share/moddable/xs/platforms/pico/xsPlatform.c:723:66: warning: comparison between pointer and integer\n' +
    '  723 |  return the->DEBUG_LOOP && the->connection && (kSerialConnection != the->connection);\n' +
    '      |                                                                  ^~\n' +
    '/var/folders/nd/2xq97b3j4fgd9zqrz1hlykvm0000gn/T//ccxgp5iL.s: Assembler messages:\n' +
    '/var/folders/nd/2xq97b3j4fgd9zqrz1hlykvm0000gn/T//ccxgp5iL.s:457: Warning: setting incorrect section attributes for .data.xsram\n' +
    '/var/folders/nd/2xq97b3j4fgd9zqrz1hlykvm0000gn/T//ccC4rD4c.s: Assembler messages:\n' +
    '/var/folders/nd/2xq97b3j4fgd9zqrz1hlykvm0000gn/T//ccC4rD4c.s:2846: Warning: setting incorrect section attributes for .rodata',
  failed: true,
  timedOut: false,
  isCanceled: false,
  killed: false
}

Node.js v18.12.1
```

</details>

It may be that the Pico setup code in xs-dev needs to be updated. The dependency on `pioasm` is relatively new, arriving when Pico W platform support was implemented. @mkellner maybe able to provide some information to help with this.

mkellner, 2022-11-24:
`pioasm` is built as a part of the pico tools.

```
cd $PICO_SDK_DIR
mkdir build
cd build
cmake ..
make
```

phoddie, 2022-11-24:
So maybe xs-dev is pinned to an older version of the Pico tools so it did t get that? When I ran that install it definitely spent a bunch of time building the Pico tools.

PR #87 (2022-12-20): fix: add env checks before attempting to run project code
Fixes #78 
Fixes #79 
Fixes #72
closed 2022-12-20 by HipsterBrown