HipsterBrown/xs-dev #96: Extra step on installing ESP32 SDK with xs-dev

rei-vilo, 2023-01-23:
Installing the ESP32 SDK through `xs-dev setup --device=esp32` requires an extra step.

Environment is a MacBook Air M1 2020 running on macOS 13.1 (22C65).

```
⠧ Installing esp-idf tooling
All done! You can now run:

  . ./export.sh

✔ Installing esp-idf tooling
ℹ Sourcing esp-idf environment
✔ 
  Successfully set up esp32 platform support for Moddable!
  Test out the setup by starting a new terminal session, plugging in your device, and running: xs-dev run --example helloworld --device=esp32
  If there is trouble finding the correct port, pass the "--port" flag to the above command with the path to the /dev.cu.* that matches your device.
```

Problem is, the correct folder for running `. ./export.sh` is not mentioned .

Correct procedure is

```
cd /Users/USER/.local/share/esp32/esp-idf
. ./export.sh
```

Could `xs-dev` manage this extra step? 

Thank you!

phoddie, 2023-01-23:
FWIW – [Moddable SDK 3.5.0](https://github.com/Moddable-OpenSource/moddable/releases/tag/3.5.0) and later takes care of that step automatically. That's mentioned in the release notes:

> ESP32 builds in environments that have not sourced $IDF_PATH/exports.sh now automatically source it

That should eliminate the need for xs-dev to perform this step.

rei-vilo, 2023-01-23:
Thank you for your answers. 

It is a good idea to silence the installer, as there is (too) much documentation to read!

closed 2023-01-23 by rei-vilo