HipsterBrown/xs-dev #101: command line configuration arguments

phoddie, 2023-02-08:
When running network examples on a device, it is necessary to set the Wi-Fi credentials. This is most commonly done by passing the credentials on the command line arguments:

```
cd $MODDABLE/examples/network/http/httpget
mcconfig -d -m -p esp32 ssid=MyWiFi password="a secret"
```

This adds `ssid` and `password` to the properties of the `mc/config` module, which is automatically used at start-up to connect to Wi-Fi.

xs-dev does not currently use such command line arguments, not does it generate an error when they are present:

```
xs-dev run --example network/http/httpsget ssid=MyWiFi password="a secret"
```

This feature is used in other places in the Moddable, for example to configure the touch driver. It is also used by projects to change their behavior. It would be great to see it supported in xs-dev as well, using the same syntax if practical.

HipsterBrown, 2023-02-09:
I knew this aspect of the mcconfig would eventually come up 😅 

I enjoy using flags to be explicit about arguments passed to a CLI process, so how does the following look to you?

```sh
xs-dev run --example network/http/httpsget --config ssid=MyWiFi,password="a secret"
```

Or even something like this (probably my preference):

```sh
xs-dev run --example network/http/httpsget --config.ssid MyWiFi --config.password "a secret"
```

In both cases, it makes the setting of the ssid and password fields of the project config clear. I can handle both use cases in the code. 

---

Aside from the `mc/config` setting for network examples, I know xs-dev does not support a specific `screen` or `touch` configuration argument at the moment either. This could also be accomplished through the `--config` flag to stay consistent with the way mcconfig currently works, or be separate flags in their own right, i.e. `--screen` or `--touch`.

phoddie, 2023-02-09:
@HipsterBrown – My preference might be a blend: `--config.ssid=MyWiFi`. There are a couple reasons for that. Using `config.ssid` matches how the property will be accessed from script. It even lends itself to things like `config.wifi.etc` down the road (the config itself is just JSON so nested objects are allowed, just not currently from the command line). The form `ssid=MyWiFi` using an equal sign instead of a space matches what `mcconfig` accepts, which is less to remember for those of us using both.

Whatever approach `xs-dev` takes here, I think it should reject the `mcconfig` approach rather than failing silently, so users have one less mistake to puzzle over.

I don't see a strong reason to do anything special for `touch` and `screen`. `mcconfig` doesn't (at least not from the perspective of  command line arguments). They are useful, but still obscure enough that they don't really need their own flags.

HipsterBrown, 2023-02-23:
Resolved by v0.22.0 of the CLI

closed 2023-02-23 by HipsterBrown