HipsterBrown/xs-dev #98: xs-dev scan - can't find esptool.py

phoddie, 2023-02-08:
The `scan` feature of xs-dev is incredibly useful for checking that devices are correct connected. Recent changes to the Moddable SDK are causing it to fail:

```
> xs-dev scan
esptool.py required to scan for Espressif devices. Setup environment for ESP8266 or ESP32:
 xs-dev setup --device esp32
 xs-dev setup --device esp8266.
⚠ No available devices found.
```

Before the changes, the Moddable SDK required the user to `source $IDF_PATH/exports.sh` before running any ESP32-related commands. Now that is taken care of automatically by the build. As a result, `esptool.py` isn't available by default when `xs-dev` runs. The text above is misleading, because it suggests that the ESP32 tools aren't installed, though they are.

At a minimum, it seems like the text above could be updated so that when $IDF_PATH is defined, a more accurate message is disabled. Perhaps it would make sense to perform the `source` automatically in that case, so the user doesn't have to. I'm unsure which approach makes more sense here.

HipsterBrown, 2023-02-09:
> Before the changes, the Moddable SDK required the user to source $IDF_PATH/exports.sh before running any ESP32-related commands. Now that is taken care of automatically by the build. As a result, esptool.py isn't available by default when xs-dev runs.

@phoddie I'm not sure I completely understand this. `xs-dev` calls out to `esptool.py` directly when looking for connected  Espressif-based devices: https://github.com/HipsterBrown/xs-dev/blob/6535899c7928cbb1ac092aa603417d748a9fa054/src/commands/scan.ts#L63

So `esptool.py` should be available in the PATH in order for the `scan` command to work, regardless of the Moddable SDK setup. 

> At a minimum, it seems like the text above could be updated so that when $IDF_PATH is defined, a more accurate message is disabled. Perhaps it would make sense to perform the source automatically in that case, so the user doesn't have to. I'm unsure which approach makes more sense here.

Attempting to source the `$IDF_PATH`, if available, makes sense to me, so I'll look into adding that check into the CLI.

phoddie, 2023-02-09:
@HipsterBrown, apologies for being obscure.

The Moddable SDK used to require the user to `source $IDF_PATH/exports.sh` themselves. There were two approaches, each with problems:
1. Adding this to a start-up shell script. This was annoying because it takes some time execute and generates a bunch of output, even in terminal windows that aren't going to do an ESP32 build. Also, some environments don't run the start-up shell script. And... because of differing shells, there is no single way to do this... which makes for complicated instructions which leads to more failures.
2. Performing the operation manually before building. The problem: people forget.
So... we automated that step. That solves the problems, but it means that `esptool.py` (and other, more obscure, IDF tools) aren't in `$PATH` now. For most developers, that is not a problem since they build and run through `mcconfig`.

> Attempting to source the $IDF_PATH, if available, makes sense to me, so I'll look into adding that check into the CLI.

Perfect. Thank you.

PR #171 (2024-08-14): fix(scan): source esp_idf before checking for esptool.py
Fixes #98 

Also addresses introduced by #170 with the removal of `source $IDF_PATH/exports.sh` from the `xs-dev-exports.sh`.
closed 2024-08-14 by HipsterBrown