HipsterBrown/xs-dev #107: set-up for xsbug console log

phoddie, 2023-03-05:
Starting in [Moddable SDK 3.6.0](https://github.com/Moddable-OpenSource/moddable/releases/tag/3.6.0) `mcconfig` implements the `-l` command line option to redirect debugging output to the console. This is necessary for headless development machines and convenient for developers who prefer to run without a debugger (I don't understand why, but I suppose it is familiar because of Node.js). This feature is implemented as [scripts](https://github.com/Moddable-OpenSource/moddable/tree/public/tools/xsbug-log) that run under Node.js. Before using the log feature, developers must first do an `npm install` to get the dependencies as [documented](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/tools/tools.md#arguments) under `mcconfig`:

```
cd $MODDABLE/tools/xsbug-log
npm install
```

Since xs-dev is already taking care of setting-up just about everything else and since it is running in Node.js we know `npm` is available, what do you think about also setting-up the log feature?

HipsterBrown, 2023-03-06:
@phoddie I think that's a great idea! 

Should the `-l` CLI flag be a default value when running `xs-dev run`? I'm thinking of it like `xs-dev run --log false` would turn off that behavior.

phoddie, 2023-03-06:
> I think that's a great idea!

Cool, thanks!

> I'm thinking of it like xs-dev run --log false would turn off that behavior.

Agreed. It would be great for `xs-dev run` to support this as an option.

> Should the -l CLI flag be a default value when running xs-dev run?

I don't think xs-dev should default to passing `-l` to `mcconfig`. Developers usually run code to debug it. They should have a real debugger by default.

closed 2023-03-07 by HipsterBrown

phoddie, 2023-03-07:
That was quick! Thank you. I'll give it a try.

HipsterBrown, 2023-03-07:
@phoddie You'll need to run setup again in order for the `npm install` step to happen before trying the new flag. Not sure how well this works on Windows yet.

phoddie, 2023-03-07:
@HipsterBrown – works as advertised (at least on macOS). Very nice. Thanks again.