HipsterBrown/xs-dev #114: mcconfig build output parameter is empty since 0.24.0 (macOS)

mrpeu, 2023-03-23:
I am getting an error on build with `0.24.0`

```
...
  Error: Command failed with exit code 1: mcconfig -m -p esp32/m5paper -t build -o  -d
...
  ### Error: -o '-d': directory not found!
...
```

I installed each intermediate version from the latest `0.24.3` down to `0.23.0`.
The first time I installed xs-dev might have been between `0.22.0` and `0.23.0`.

I must might ruined something while trying to fix this problem.
`0.23.0` used to work. It does not anymore but it's another problem I guess.
```
Command failed with exit code 2: mcconfig -m -p esp32/m5paper -t build -o /Users/XS/projects/xs/moddable/build -d
...
Traceback (most recent call last):
File "/Users/XS/.local/share/esp32/esp-idf/tools/idf.py", line 36, in <module>
  from idf_py_actions.tools import executable_exists, idf_version, merge_action_lists, realpath  # noqa: E402
File "/Users/XS/.local/share/esp32/esp-idf/tools/idf_py_actions/tools.py", line 7, in <module>
  import click
  ImportError: No module named click
    make: *** [/Users/XS/projects/xs/moddable/build/tmp/esp32/m5paper/debug/m5paper/xsProj-esp32/build/config/sdkconfig.h]
Error 1
```

xs-dev info:
```
  CLI Version                0.24.3                                                          
  OS                         Darwin                                                          
  Arch                       arm64                                                           
  Shell                      /bin/zsh                                                        
  NodeJS Version             v19.1.0 (/Users/gpe/.nvm/versions/node/v19.1.0/bin/node)        
  Python Version              (/Library/Frameworks/Python.framework/Versions/2.7/bin/python) 
  Moddable SDK Version       3.7.6 (/Users/gpe/projects/experiments/perso/moddable)          
  Supported target devices   mac, esp32                                                      
  ESP32 IDF Directory        /Users/gpe/.local/share/esp32/esp-idf
```

I'll try to un/reinstall everything now.

mrpeu, 2023-03-23:
I'm getting errors that might be unrelated when reinstalling esp-idf stuff (`xs-dev setup --device esp32`)

I'm blaming my lack of experience with Python environment [😉](https://twitter.com/phoddie/status/1631376315772436480)

[image]

closed 2023-03-23 by mrpeu

HipsterBrown, 2023-03-23:
@mrpeu I'll still look into the empty output flag getting passed to `mcconfig`. The Python dependencies are probably the biggest point of friction with the rest of the set up current, especially since esp-idf likes to ship its own version of Python and NodeJS. 

I recently started using `asdf` plugins via [`rtx`](https://github.com/jdxcode/rtx) to manage my language tooling, including [`asdf-python`](https://github.com/asdf-community/asdf-python). Your mileage may vary 😄

mrpeu, 2023-03-23:
Seeing this [issue](https://github.com/espressif/esp-idf/issues/10116), `brew unlink python` seemed to have solved the issue.
I could reliably run `xs-dev run --device simulator/m5paper` and `xs-dev run --device esp32/m5paper`.
But then `xs-dev build --deploy --device esp32/m5paper` gave the original error of this post (with the missing output parameter). And now `run` commands trigger the exception `ImportError: No module named click` again.

So it might be an issue with xs-dev, or the tools it automates of course.

Thanks for `xs-dev` anyway, the effort is grandiose.

mrpeu, 2023-03-24:
Thanks. The rest is confirmed to be about Python setups.