HipsterBrown/xs-dev #125: python 3.11 might not compile esp32 sdk

reconbot, 2023-06-03:
While installing with `xs-dev setup --device esp32` I kept getting an error. I switched from python 3.11 to 3.9 and it worked. 

With 3.11 I got a [pretty long error](https://gist.github.com/reconbot/4ee046f629ac34e5ff5f218877fa1d42) but I think this is the relevant part.

```
      building 'gevent.__greenlet_primitives' extension
      clang -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -I/opt/homebrew/opt/python@3.11/Frameworks/Python.framework/Versions/3.11/include/python3.11 -I/opt/homebrew/Cellar/python@3.11/3.11.3/Frameworks/Python.framework/Versions/3.11/include/python3.11 -I/private/var/folders/jw/9_9ff3n94n38xmfp12sr22r80000gn/T/pip-install-o7_qtajn/gevent_a61d1e701b344da085930b4351b330d8/deps -Isrc/gevent -Isrc/gevent/libev -Isrc/gevent/resolver -I. -I/Users/wizard/.espressif/python_env/idf4.4_py3.11_env/include -I/opt/homebrew/opt/python@3.11/Frameworks/Python.framework/Versions/3.11/include/python3.11 -c src/gevent/_greenlet_primitives.c -o build/temp.macosx-13-arm64-cpython-311/src/gevent/_greenlet_primitives.o -Wno-unreachable-code -Wno-deprecated-declarations -Wno-incompatible-sysroot -Wno-tautological-compare -Wno-implicit-function-declaration
      src/gevent/_greenlet_primitives.c:216:12: fatal error: 'longintrepr.h' file not found
        #include "longintrepr.h"
                 ^~~~~~~~~~~~~~~
      1 error generated.
      error: command '/usr/bin/clang' failed with exit code 1
```

I was able to switch my homebrew python version with 
```
brew unlink python@3.11 && brew link python@3.9 --overwrite
```

reconbot, 2023-06-03:
I just learned about the doctor command

This is my working environment

```
xs-dev environment info:
  CLI Version                0.25.4
  OS                         Darwin
  Arch                       arm64
  Shell                      /bin/zsh
  NodeJS Version             v18.3.0 (/Users/wizard/.local/share/rtx/installs/nodejs/18.3.0/bin/node)
  Python Version             3.9.6 (/Users/wizard/.espressif/python_env/idf4.4_py3.9_env/bin/python)
  Moddable SDK Version       3.9.2 (/Users/wizard/src/moddable)
  Supported target devices   mac, esp32
  ESP32 IDF Directory        /Users/wizard/.local/share/esp32/esp-idf
```

phoddie, 2023-06-05:
Yea, I think I bumped into the same problem. @HipsterBrown is there anything we can do to help others avoid this same problem?

HipsterBrown, 2023-06-05:
Besides setting a specific version of python to be installed via homebrew through xs-dev, we might be able to detect the python version before the homebrew step and see if it changes. If it does, we could call the unlink command automatically (or after prompting the user). Espressif also likes to include their own Python binary, as demonstrated by @reconbot's `xs-dev doctor` output. 

The least we can do is provide better error handling in the `run`/`build` workflow and document this in the troubleshooting section of the xs-dev.js.org site.

reconbot, 2023-06-05:
I think you can set the path to the right version of python for the sub process of the tools. I wouldn't mess with linking.