HipsterBrown/xs-dev #126: typescript has to be installed globally and have tsc in path

reconbot, 2023-06-03:
When using a typescript project, typescript has to be installed globally and your shell has to have tsc in it's path. This isn't mentioned in the docs or via the doctor command.

phoddie, 2023-06-04:
(These notes are more for Nick than Francis, but more voices on the optimal behavior here can only help.)

@reconbot raises a good point. The TypeScript compiler is an  example of a dependency that applies to some Moddable SDK projects, but not all. The fontbm tool is another. We don't want to require every developer to install these tools, but we don't want developers to puzzle through build failures when they are missing.

In theory recursively processing the project manifest can identify these dependencies. Processing the manifest is an adventure. I recently updated mcconfig to output a flat version of the project manifest to `manifest_flat.json`. That is a start but not enough to easily know, for example, if there are TypeScript files or use of fontbm.

I did some more hacking on mcconfig tonight to add that information to `manifest_flat.json`. Just by checking `manifest.["x-files-info"].ts.length`  a tool can know if aa project uses TypeScript. Here's a simple example of output for helloworld on macOS with main.js changed to main.ts and manifest_typings.json included:

<details>
<summary>manifest_flat.json</summary>

```json
{
	"config": {
		"format": "RGB565LE",
		"rotation": 0
	},
	"creation": {
		"keys": {
			"initial": 32,
			"incremental": 32,
			"name": 53,
			"symbol": 3
		},
		"parser": {
			"buffer": 1024,
			"table": 17
		},
		"main": "main",
		"chunk": {
			"initial": 32768,
			"incremental": 1024
		},
		"heap": {
			"initial": 2048,
			"incremental": 64
		},
		"stack": 384,
		"static": 0
	},
	"defines": {
		
	},
	"data": {
		
	},
	"modules": {
		"*": [
			"/Users/hoddie/Projects/moddable/build/simulators/modules/screen",
			"/Users/hoddie/Projects/moddable/modules/base/time/*",
			"/Users/hoddie/Projects/moddable/modules/base/time/mac/*",
			"/Users/hoddie/Projects/moddable/modules/base/timer/*",
			"/Users/hoddie/Projects/moddable/modules/base/timer/mac/*",
			"/Users/hoddie/Projects/moddable/modules/base/timer/modTimer",
			"/Users/hoddie/Projects/moddable/modules/files/resource/*",
			"/Users/hoddie/Projects/moddable/modules/base/instrumentation/modInstrumentation",
			"/Users/hoddie/Projects/moddable/typings/*",
			"/Users/hoddie/Projects/moddable/examples/helloworld/main"
		],
		"commodetto/*": [
			"/Users/hoddie/Projects/moddable/typings/commodetto/*"
		],
		"dns/*": [
			"/Users/hoddie/Projects/moddable/typings/dns/*"
		],
		"mc/*": [
			"/Users/hoddie/Projects/moddable/typings/mc/*"
		],
		"pins/*": [
			"/Users/hoddie/Projects/moddable/typings/pins/*"
[... 513 lines cut ...]
				bleServices: this.bleServicesFiles.slice(),
//				bmpAlpha: this.bmpAlphaFiles.slice(),
//				bmpColor: this.bmpColorFiles.slice(),
				bmpFont: this.bmpFontFiles.slice(),
				bmpMask: this.bmpMaskFiles.slice(),
				c: this.cFiles.slice(),
				cdv: this.cdvFiles.slice(),
				clut: this.clutFiles.slice(),
				data: this.dataFiles.slice(),
				dts: this.dtsFiles.slice(),
				h: this.hFiles.slice(),
				image: this.imageFiles.slice(),
				js: this.jsFiles.slice(),
				nodered2mcu: this.nodered2mcuFiles.slice(),
				resources: this.resourcesFiles.slice(),
				sound: this.soundFiles.slice(),
				string: this.stringFiles.slice(),
				ts: this.tsFiles.slice(),
			}
		};

		this.writeFileString(this.tmpPath + this.slash + "manifest_flat.json", JSON.stringify(flat, undefined, "\t")); 
```

</summary>

HipsterBrown, 2023-06-05:
@phoddie Does the `typescript.compiler` field refer to the binary that is called when processing the project's TypeScript files?  Could that be a path to a binary within the project's `node_modules` directory?

> With this in place, xs-dev could use mcconfig -d (without the -m) to process the manifest. It can check the dependencies.

I wonder if there is also a path towards making `mcconfig.js` an import-able ES module for xs-dev when available, rather than shelling out for this info. That is probably a longer term refactor and collaboration, but I want to surface the idea for future conversations around a smoother developer experience. 

For the `manifest_flat.json` approach, xs-dev could hash & cache that file to make subsequent runs quicker when the project manifest doesn't change.

phoddie, 2023-06-05:
> Does the typescript.compiler field refer to the binary that is called when processing the project's TypeScript files? Could that be a path to a binary within the project's node_modules directory? 

Yes, that [appears to be the case](https://github.com/Moddable-OpenSource/moddable/blob/24d84d7b340c6bfda73b3b1499ebf88d59aaeec8/tools/mcmanifest.js#L618
).

> I wonder if there is also a path towards making mcconfig.js an import-able ES module for xs-dev when available, rather than shelling out for this info. That is probably a longer term refactor and [...]

Should be possible. I'm open to exploring that. The implementation of mcconfig / mcrun is somewhat tightly coupled to XS, however, as it relies on the native tools module. The same could be done in Node too. Still, mcconfig runs very quickly (the build takes time, but the processing is nearly instant) so performance isn't a clear motivation.

For the manifest_flat.json approach, xs-dev could hash & cache that file to make subsequent runs quicker when the project manifest doesn't change.

Very true. That would certainly let you avoid unnecessary checks in the environment (e.g. is `tsc`  available). But it is the generated makefile that actually decides which files have changed and, consequently, what needs to be rebuilt.