HipsterBrown/xs-dev #130: docs: wasm simulator playground

HipsterBrown, 2023-06-17:
It is possible to [target wasm](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/devices/wasm.md) as a platform for the build/run commands with xs-dev and open the simulator in a browser by serving the output via a static file server. With this in mind, could it be feasible to create an online "playground" (see [TypeScript](https://www.typescriptlang.org/play), [Docker](https://labs.play-with-docker.com), [Pebble](http://pebble.github.io/rockyjs/examples/simple/index.html)), that provides example programs running in the hosted simulator?

As the capabilities of the wasm simulator are expanded to include sensor data from the browser, simulated files, etc, then there might be a path to compiling user input as a "mod" to run in the wasm host program. This could also lead to flashing these examples to a local device, as demonstrated by [esptool-js](https://espressif.github.io/esptool-js/).

dashcraft, 2023-06-17:
I've been too busy to devote much time to think through the hot reloading architecture, but this reminds me of 
https://wokwi.com/projects/322410731508073042

The esp32 in browser simulator we came across awhile back. They have a web component element kit, https://github.com/wokwi/wokwi-elements, that could probably be leveraged for open source stuff as well.

HipsterBrown, 2023-06-17:
Oh that's a very cool project and very much what I'm imagining for this feature. They appear to have a rp2040 simulator as well https://github.com/wokwi/rp2040js

I have not tried extending the xs-sim functionality before but these could be very useful for the wasm target at least.