HipsterBrown/xs-dev #143: Preparing for ESP-IDF 5 support

phoddie, 2023-08-29:
The Moddable SDK will move to ESP-IDF 5 in the near future (probably not for the September release, but should be by October). We're currently targeting ESP-IDF v5.1.1.

The `target-branch ` option to xs-dev's setup " makes it easy to get the Moddable SDK `idf-v5` branch to try that out. The ESP32 set-up, however, is still pinned to v4.4.3....

https://github.com/HipsterBrown/xs-dev/blob/b6357ec7e482f3fbf9eb3870872962aae064880a/src/toolbox/setup/esp32.ts#L16

...so the build will fail.

We've discussion having xs-dev get version dependencies from the Moddable SDK. I'm not sure how practical that is to do. But, this is a good test case. ;)

Worst case, we can synchronize on the release of the Moddable SDK that transitions to ESP-IDF 5 an update xs-dev at about the same time.

HipsterBrown, 2023-08-30:
@phoddie is there an expected timeline for the SDK release to support v5 IDF? 

If there's time, I'd like to continue the work on the "platform.json" configuration files support to enable SDK-driven automated setup in xs-dev. 

Otherwise, we can add some conditional logic in the ESP32 setup & run code paths to ensure the correct version of the IDF is used based on the Moddable SDK release version.

phoddie, 2023-08-30:
> ...is there an expected timeline for the SDK release to support v5 IDF?

The to-do list keeps getting shorter. it looks to be on-track for the October release, so there's some time. 

> If there's time, I'd like to continue the work on the "platform.json" configuration files support to enable SDK-driven automated setup in xs-dev.

That sound great. If you have something to try, we can put that to the public Moddable repo to try out ahead of a formal release.

FWIW – I'm still not sure how you'll use "platform.json" for this. I'm happy to follow your lead to explore that since xs-dev is the first consumer of the information.

phoddie, 2023-09-12:
Quick update. IDF v5 support is feature complete and solid. We plan to merge that to our public main branch in the next week or two, so it will have a little time to settle ahead of the October release.

HipsterBrown, 2023-10-12:
Better late than never on my part 😅 This has been resolved in the latest release v0.28.0. 🚀

closed 2023-10-12 by HipsterBrown