HipsterBrown/xs-dev #144: windows setup for nrf52 fails

phoddie, 2023-09-26:
I'm setting up a new Windows machine and using xs-dev to get the Moddable SDK installed. The initial setup worked smoothly and the examples run as expected. Very nice.

Setting for nrf52 fails as shown below. My guess is that this is related to the xz lib issue mentioned in [trouble shooting](https://xs-dev.js.org/troubleshooting/)  but I didn't see instructions for how to resolve that on Windows.

```
> xs-dev setup --device nrf52
i Ensuring nrf52 directory
- Downloading GNU Arm Embedded ToolchainC:\Users\Peter\AppData\Roaming\npm\node_modules\xs-dev\node_modules\gluegun\build\index.js:15
    throw up;
    ^

TypeError: createUnxz is not a function
    at Object.default_1 [as nrf52] (C:\Users\Peter\AppData\Roaming\npm\node_modules\xs-dev\build\toolbox\setup\nrf52.js:93:22)
    at async Command.run (C:\Users\Peter\AppData\Roaming\npm\node_modules\xs-dev\build\commands\setup.js:66:13)

Node.js v18.18.0
```

closed 2023-09-29 by HipsterBrown

phoddie, 2023-09-29:
This error is resolved for me after updating xs-dev. Thank you!

The build runs almost to completion.  At the very end it fails with a Python (surprise!) error.

```
creating xs_nrf52.out
link to .out file

# Version
#  XS:    4.1
# Memory Usage
#  Ram:    41052 bytes
#  Heap:  217088 bytes
#  Flash:  529664 bytes
#  QSPI:          bytes

Making: C:\Users\Peter\xs-dev\moddable\build\bin\nrf52\moddable_four\debug\helloworld\xs_nrf52.uf2 from xs_nrf52.hex
Python was not found; run without arguments to install from the Microsoft Store, or disable this shortcut from Settings > Manage App Execution Aliases.
NMAKE : fatal error U1077: 'python C:\Users\Peter\xs-dev\nrf52\uf2conv.py C:\Users\Peter\xs-dev\moddable\build\bin\nrf52\moddable_four\debug\helloworld\xs_nrf52.hex -c -f 0xADA52840 -o C:\Users\Peter\xs-dev\moddable\build\bin\nrf52\moddable_four\debug\helloworld\xs_nrf52.uf2' : return code '0x2331'
Stop.
```

When the nRF52 setup ran, it appears to install Python without error. Here's a re-run of the command:

```
>xs-dev setup --device nrf52
i Ensuring nrf52 directory
| Installing python from winget
>
```

This may be another symptom of #145. Do you think it is worth opening this as a separate issue or just wait for some progress on #145 first?

phoddie, 2023-09-29:
Brief update. I manually installed Python 3.11 from the Microsoft Store. That allows the nRF52 build to complete. The binary installs and debugging works on Moddable Four. Success!

It also eliminates the warning about Python being missing when launching the Moddable Console. I tried an ESP32 build. That got further but failed here:

```
Reconfiguring ESP-IDF...
Traceback (most recent call last):
  File "C:\Users\Peter\xs-dev\esp32\esp-idf\tools\idf.py", line 36, in <module>
    from idf_py_actions.tools import executable_exists, idf_version, merge_action_lists, realpath  # noqa: E402
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Peter\xs-dev\esp32\esp-idf\tools\idf_py_actions\tools.py", line 7, in <module>
    import click
ModuleNotFoundError: No module named 'click'
NMAKE : fatal error U1077: 'python %IDF_PATH%\tools\idf.py -n reconfigure -D SDKCONFIG_DEFAULTS=C:/Users/Peter/xs-dev/moddable/build/tmp/esp32/moddable_two/debug/helloworld/xsProj-esp32/sdkconfig.mc -D SDKCONFIG_HEADER="C:\Users\Peter\xs-dev\moddable\build\tmp\esp32\moddable_two\debug\helloworld\xsProj-esp32\build\config\sdkconfig.h" -D CMAKE_MESSAGE_LOG_LEVEL=ERROR -D DEBUGGER_SPEED=460800 -D IDF_TARGET=esp32 -D ESP32_SUBCLASS=esp32 -D SDKCONFIG_DEFAULTS=C:\Users\Peter\xs-dev\moddable\build\tmp\esp32\moddable_two\debug\helloworld\xsProj-esp32\sdkconfig.mc' : return code '0x1'
Stop.
```

I've seen that failure on "click" before on macOS. I think it has something to do with a Python version mismatch?

HipsterBrown, 2023-09-29:
> I've seen that failure on "click" before on macOS. I think it has something to do with a Python version mismatch?

Not exactly the same error but a similar effect related to newer versions of Python not working with the ESP-IDF 😞  https://github.com/HipsterBrown/xs-dev/issues/125

phoddie, 2023-09-29:
Ah, right. the ESP-IDF docs suggest 3.8 or later should work. But... I don't think IDF should be using the system Python anyway, so the problem is likely more about why it isn't using the ESP-IDF Python.

HipsterBrown, 2023-09-30:
@phoddie I was reading through your update on the nrf52 build again and noticed you mentioned using Python 3.11 to successfully compile for the Moddable Four on Windows. Does that mean Python 2.7 is not actually required as documented in the Moddable repo and codified in xs-dev? https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/devices/nrf52.md#installing-1

phoddie, 2023-09-30:
I believe that is correct. The dependency on Python 2.7 may only be for ESP8266 builds. On macOS, I'm using Python 3.10.10 for nRF52 builds.

PR #147 (2023-09-29): fix(setup/nrf52): create the decompression stream when needed
Fixes #144 
Fixes #146