HipsterBrown/xs-dev #145: windows setup for ESP32 fails - python not found

phoddie, 2023-09-26:
I ran xs-dev setup for ESP32 and it reported success. With a fresh Moddable command prompt, the doctor command shows that the ESP-IDF is there. 
```
>xs-dev doctor
xs-dev environment info:
  CLI Version                0.27.4
  OS                         Windows_NT
  Arch                       x64
  Shell                      Unknown
  NodeJS Version             v18.18.0 (C:\Program Files\nodejs\node.EXE)
  Python Version             Unavailable (C:\Users\Peter\xs-dev\moddable.BAT)
  Moddable SDK Version       4.1 (C:\Users\Peter\xs-dev\moddable)
  Supported target devices   win, esp32
  ESP32 IDF Directory        C:\Users\Peter\xs-dev\esp32\esp-idf

If this is related to an error when using the CLI, please create an issue at "https://github.com/hipsterbrown/xs-dev/issues/new" with the above info.
```

However, it fails to build helloworld because it cannot find Python.

```
>xs-dev run --example helloworld --device esp32
\ Building and deploying project C:\Users\Peter\xs-dev\moddable\examples\helloworld on esp32
| Building and deploying project C:\Users\Peter\xs-dev\moddable\examples\helloworld on esp32
Using Python in C:\Users\Peter\xs-dev\esp32\esp-idf\
'null' is not recognized as an internal or external command,
operable program or batch file.
Using Git in C:\Espressif\tools\idf-git\2.34.2\cmd\
/ Building and deploying project C:\Users\Peter\xs-dev\moddable\examples\helloworld on esp32

Error: The following tools are not installed in your environment.

  python

Please use the Windows Tool installer for setting up your environment.
Download link: https://dl.espressif.com/dl/esp-idf/
For more details please visit our website: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/windo\ Building and deploying project C:\Users\Peter\xs-dev\moddable\examples\helloworld on esp32
Python was not found; run without arguments to install from the Microsoft Store, or disable this shortcut from Settings 
```

Running python without any arguments launches the Windows Store to install it. I didn't try that as I'm pretty sure that the ESP-IDF already includes its preferred Python version.

phoddie, 2023-09-26:
When launching the Moddable command prompt, the following message is displayed. Just FYI.

```
**********************************************************************
** Visual Studio 2022 Developer Command Prompt v17.7.4
** Copyright (c) 2022 Microsoft Corporation
**********************************************************************
[vcvarsall.bat] Environment initialized for: 'x86'
Setting PYTHONNOUSERSITE, was not set
Using Python in C:\Users\Peter\xs-dev\esp32\esp-idf\
'null' is not recognized as an internal or external command,
operable program or batch file.
Using Git in C:\Espressif\tools\idf-git\2.34.2\cmd\
git version 2.34.1.windows.1

Error: The following tools are not installed in your environment.

  python

```

HipsterBrown, 2023-09-27:
I see that the Python version is not found when running `xs-dev doctor` in the fresh Moddable command prompt: `Python Version             Unavailable (C:\Users\Peter\xs-dev\moddable.BAT)`

It's my understanding that the ESP-IDF should be providing its vendored version of Python as well. The message when running the hello world example appears to come from the esp-idf script to determine which Python version to use:

```
Using Python in C:\Users\Peter\xs-dev\esp32\esp-idf\
'null' is not recognized as an internal or external command,
operable program or batch file.
```

The code for calling the esp-idf installer on Windows can be found here: https://github.com/HipsterBrown/xs-dev/blob/main/src/toolbox/setup/esp32/windows.ts#L11

cc @andycarle

andycarle, 2023-09-28:
Curious. I'll be able to take a solid look at this early next week.

andycarle, 2023-10-03:
So far I'm not able to reproduce this. But I have a decent idea of how it could happen. 

When I start fresh with `xs-dev setup` and `xs-dev setup --device=esp32`, I get this from `xs-dev doctor`:

```
xs-dev environment info:
  CLI Version                0.27.5
  OS                         Windows_NT
  Arch                       x64
  Shell                      Unknown
  NodeJS Version             v18.16.1 (C:\Program Files\nodejs\node.EXE)
  Python Version             3.8.7 (C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\python.EXE)
  Moddable SDK Version       4.2.0 (C:\Users\andyc\xs-dev\moddable)
  Supported target devices   win, esp32
  ESP32 IDF Directory        C:\Users\andyc\xs-dev\esp32\esp-idf
```

My bet is that the Espressif Tools installer did not successfully run on @phoddie's system. That would cause the Espressif Python virtual environment to be missing when the ESP-IDF environment batch file runs. 

I'm going to try to get my system to a cleaner state (without those tools pre-existing) to chase this further.

andycarle, 2023-10-03:
@phoddie Do you happen to recall if this part of the process completed successfully on your machine:

[image]

I'm seeing some issues on my machine with the Espressif installer. Basically, it is having trouble overwriting my old version that I "removed" (apparently incompletely) with the Espressif uninstaller.