HipsterBrown/xs-dev #148: feat(setup): create python virtual environment for isolated dependencies and tooling

HipsterBrown, 2023-09-30:
As I've been working in Python more recently, [virtual environments](https://docs.python.org/3/library/venv.html) appear to be a best practice to ensure Python packages and tools don't conflict with the global dev environment. Given some of the recent issues related to Python (#145, #125), it might help to manage this environment on behalf of xs-dev users with an option to opt-out of this behavior.

The expected workflow would be to create a virtual environment once Python is determined to be required as part of the setup process, which appears to be the esp8266, esp32, and nrf52 setups. That virtual environment would be stored in `~/.local/share/xs-dev/.venv`. This venv will be activated by xs-dev before installing any Python packages or executing related scripts like `esp-idf/export.sh`.

This could potentially lead to a separate environment for each required device to prevent clashing requirements on systems that develop against multiple target hardware. 

As an alternative (or in addition) to this, the xs-dev docs can cover this topic as it relates to the various embedded tooling. I don't want xs-dev to become a Python version or environment manager as much as it can be helped. 🤞

phoddie, 2023-10-03:
> I don't want xs-dev to become a Python version or environment manager as much as it can be helped.

No kidding! It does kind of feel like each environment wants its particular version of Python.

HipsterBrown, 2023-10-03:
> It does kind of feel like each environment wants its particular version of Python.

The first step is just documenting this knowledge as part of the setup and/or troubleshooting pages. Then look into automating the concerns through xs-dev without making it impossible to debug. 😅

andycarle, 2023-10-03:
I'm not opposed to this idea. But I do think we want Espressif's tools to manage their own Python virtual environment for the ESP32. The issue in #145 appears to me to be a failure during the Espressif installation process.

HipsterBrown, 2024-08-28:
Closing this in favor of the approach implemented in #172

closed 2024-08-28 by HipsterBrown