HipsterBrown/xs-dev #151: Device setup error on RPi5: python-setuptools has no installation candidate

intGus, 2023-12-08:
I get this error on a Raspberry Pi 5 with Bookworm when trying to setup a esp32 device:

  exitCode: 100,
  signal: undefined,
  signalDescription: undefined,
  stdout: undefined,
  stderr: "E: Package 'python-setuptools' has no installation candidate",
  failed: true,
  timedOut: false,
  isCanceled: false,
  killed: false

I went to Moddable website and followed their instructions to install the SDK and I realised they have the same command. I think it should be python3-setuptools since I tried and installed everything correctly but xs-dev is not recognising that installation and the device setup won't continue.

intGus, 2023-12-08:
I would love to contribute and will do it in a bit. I just fixed it locally and it worked. Now I'm trying to get my esp32-s2 recognized.

HipsterBrown, 2023-12-08:
This fix has been released in version 0.28.1 🚀 Thanks for your contribution!

PR #152 (2023-12-08): fix(setup/esp32): update a package name in linux.ts
Changed the package name from python-setuptools to python3-setuptools

Fixes #151