HipsterBrown/xs-dev #159: setup fails on macOS ENOENT npm install

mrpeu, 2024-01-18:
From what I understand of this error, the tool (gluegun? cross-spawn?) does not succeed in calling `npm`. It is globally installed with `nvm` on this system and is usually working fine.

macOS Sonoma 14.2.1
node 21.6.0
npm 10.2.4
xs-dev 0.30.2

```
> xs-dev setup
(node:84812) [DEP0040] DeprecationWarning: The `punycode` module is deprecated. Please use a userland alternative instead.
(Use `node --trace-deprecation ...` to show where the warning was created)
Setting up the mac tools!
ℹ Moddable repo already installed
ℹ xsbug.app symlink already exists
⠋ Installing xsbug-log dependencies/Users/xxx/.nvm/versions/node/v21.6.0/lib/node_modules/xs-dev/node_modules/gluegun/build/index.js:14
    throw up;
    ^

Error: Command failed with ENOENT: npm install
spawn npm ENOENT
    at ChildProcess._handle.onexit (node:internal/child_process:286:19)
    at onErrorNT (node:internal/child_process:484:16)
    at process.processTicksAndRejections (node:internal/process/task_queues:82:21) {
  errno: -2,
  code: 'ENOENT',
  syscall: 'spawn npm',
  path: 'npm',
  spawnargs: [ 'install' ],
  originalMessage: 'spawn npm ENOENT',
  shortMessage: 'Command failed with ENOENT: npm install\nspawn npm ENOENT',
  command: 'npm install',
  escapedCommand: 'npm install',
  exitCode: undefined,
  signal: undefined,
  signalDescription: undefined,
  stdout: '',
  stderr: '',
  failed: true,
  timedOut: false,
  isCanceled: false,
  killed: false
}

Node.js v21.6.0
```

Has anyone any idea where it is going wrong?

HipsterBrown, 2024-01-19:
@mrpeu Did you have an existing install of Moddable on your system that was not set up by xs-dev, i.e  manually by following the Moddable docs? 

That step is looking for the xsbug-log directory within the Moddable repo on your system. 

Can you running `xs-dev info` and share the output?

mrpeu, 2024-01-19:
I did indeed install Moddable manually. What conflicts does that create?

```
> xs-dev info
(node:92606) [DEP0040] DeprecationWarning: The `punycode` module is deprecated. Please use a userland alternative instead.
(Use `node --trace-deprecation ...` to show where the warning was created)
/Users/xxx/.nvm/versions/node/v21.6.0/lib/node_modules/xs-dev/node_modules/gluegun/build/index.js:14
    throw up;
    ^

Error: Command failed: git tag -l --sort=-taggerdate
fatal: not a git repository (or any of the parent directories): .git

    at genericNodeError (node:internal/errors:984:15)
    at wrappedFn (node:internal/errors:538:14)
    at ChildProcess.exithandler (node:child_process:422:12)
    at ChildProcess.emit (node:events:519:28)
    at maybeClose (node:internal/child_process:1105:16)
    at Socket.<anonymous> (node:internal/child_process:457:11)
    at Socket.emit (node:events:519:28)
    at Pipe.<anonymous> (node:net:337:12) {
  code: 128,
  killed: false,
  signal: null,
  cmd: 'git tag -l --sort=-taggerdate',
  stdout: '',
  stderr: 'fatal: not a git repository (or any of the parent directories): .git\n'
}

Node.js v21.6.0
```

mrpeu, 2024-01-19:
Thank you @HipsterBrown
It must be installed in several different way tbh so it might be a mess if things reference each other with environment variables or default folder path.
I should cleanup everything and try again: from 0, only with `xs-dev`.
I'll delete the moddable and esp folders I can find in the user folders, and check files such as `~/.bashrc`.

`xs-dev` has been very helpful already, I understand that the magic dependencies it needs must be a mess to keep track of. Especially since the main goal of `xs-dev` is precisely to "hide" all this complexity :)
I'll come back after I tried a clean reinstall.

HipsterBrown, 2024-01-19:
I appreciate you understanding the complexity of juggling so much in a single CLI 😄. xs-dev tries to rely on the same environment variables as the Moddable SDK to ensure everything is set up for success when building programs for the various target devices, especially if the tools have been manually installed previously or while working on custom branches. 

I've just published version 0.30.3 of xs-dev to provide improved output from the `info` command based on this feedback. Overall better handling is on the roadmap to ship v1.0. Thank you for your patience and trying out xs-dev. 🙏

mrpeu, 2024-01-29:
Stale. Will reopen if still happens after a system cleanup.

closed 2024-01-29 by mrpeu

mrpeu, 2024-11-19:
Seeing the suspicious line about Moddable SDK Version "Not found (/Users/gpe/.local/share/moddable)"
[image]
This was simply fixed by removing this folder all together and re-`xs-dev setup` 👍

HipsterBrown, 2024-11-20:
That Python path is also interesting. Not sure why it didn't find Python 3.x on your system, assuming it was added by Homebrew during the setup process if it wasn't there already. 🤔

mrpeu, 2024-11-20:
Python has been installed by a colleague on this machine because I hate dealing with Python environments and didn't even know a thing about it at the time. Since then I try to touch it the least I can. There has been some Homebrew involved and some other shenanigans to make it work on M1 (arm) at the time. So, no worries: it is expected to be dirty around there.