HipsterBrown/xs-dev #163: xs-dev setup fails on Raspberry pi OS bookworm 64b

mshioji, 2024-05-04:
Hello, 
I found similar issue was created in the past https://github.com/HipsterBrown/xs-dev/issues/68 , but it seems it is closed so I think I'm trying to post new one.

OS: Raspberry pi OS bookworm 64bit full
Hardware: Raspberry Pi 4B (4GB)
*this is happening when OS upgrade is applied. somehow, setup goes well if the OS is 'NOT' updated from the Released os Image.
$ sudo apt update
$ sudo apt upgrade -y

### Try to setup Moddable-SDK with xs-dev
$ xs-dev setup

✔ Installing GTK+ 3...
ℹ Downloading release tools

/usr/lib/node_modules/xs-dev/node_modules/gluegun/build/index.js:15
    throw up;
    ^

### Try to see Moddable-SDK install with xs-dev setup --target-branch public
$ xs-dev teardown
$ xs-dev setup --target-branch public

✔ Installing GTK+ 3...
✔ Cloning https://github.com/Moddable-OpenSource/moddable repo
⠋ Building platform tooling/usr/lib/node_modules/xs-dev/node_modules/gluegun/build/index.js:15
    throw up;
    ^

### as an alternative, manually install moddable-SDK @/home/.local/share)

$ cd /home/.local/share
$ git clone https://github.com/Moddable-OpenSource/moddable
Cloning into 'moddable'...
fatal: unable to access 'https://github.com/Moddable-OpenSource/moddable/': Could not resolve host: github.com (failed first try)
$ git clone https://github.com/Moddable-OpenSource/moddable
Cloning into 'moddable'...          (it's OK when second try)

$ reboot

$ cd $MODDABLE/build/makefiles/lin
$ make
$ make install

### device setup
$ xs-dev setup --device esp32

it seems OK, helloworld and node-red-mcu also works fine but:
  *when opening new terminal window, it shows following:
Not using an unsupported version of tool openocd-esp32 found in PATH: 0.12.0. To use it, run '/usr/bin/python3 /home/mshioji/.local/share/esp32/esp-idf/tools/idf_tools.py export --prefer-system'

Thank you,

mshioji, 2024-05-05:
Hi,
I'm trying setup raspberry pi environment several times since our team going to update guidebook.
in this trial, it seems device setup also fail sometimes, with similar situation:

✔ Installing build dependencies
ℹ Configuring $IDF_PATH
⠋ Installing esp-idf tooling/usr/lib/node_modules/xs-dev/node_modules/gluegun/build/index.js:15
    throw up;
    ^

Error: Command failed with ENOENT: ./install.sh

in this case, reboot and retry "xs-dev setup --device esp32" again, it is succeeded.
I will further try...

Thank you!

HipsterBrown, 2024-05-06:
Thanks for reporting this issue @mshioji. I will try to recreate it on my own Raspberry Pi 4B and report back.

HipsterBrown, 2024-05-06:
@mshioji Are you able to share the output of `xs-dev info` from your Raspberry Pi environment? Also, can you provide some information about how you installed Node.js and the `xs-dev` CLI?

HipsterBrown, 2024-05-06:
I just installed Raspbian OS 64bit Full on a Pi 4B (4GB RAM) and was able to successfully set up the Moddable SDK and ESP32 development tooling with xs-dev. The following is a summary of commands run on the device after logging in the first time:

```console
sudo apt update
sudo apt upgrade -y
sudo apt install -y nodejs npm
sudo npm i -g xs-dev
xs-dev setup
xs-dev setup --device esp32
```

This is what `xs-dev` info output for my system after successful setup:

```console
dev@dev:~ $ xs-dev info
xs-dev environment info:
  CLI Version                0.30.4
  OS                         Linux
  Arch                       arm64
  Shell                      /bin/bash
  NodeJS Version             v18.19.0 (/usr/bin/node)
  Python Version             3.11.2 (/home/dev/.espressif/python_env/idf5.1_py3.11_env/bin/python)
  Moddable SDK Version       4.6.0 (/home/dev/.local/share/moddable)
  Supported target devices   lin, esp32
  ESP32 IDF Directory        /home/dev/.local/share/esp32/esp-idf
```

mshioji, 2024-05-07:
Hello @HipsterBrown ,
Thank you for looking at this. Now I'm out so I don't have xs-dev info right now, I will provide info later.
I tried with NodeJS v20.12.2 and npm v10.5.0 as following steps:
(I think every modules installed as latest)

---> from disk image Raspberry pi OS bookworm full 64bit
```
$sudo apt update
$sudo apt upgrade -y

$ curl -fsSL https://deb.nodesource.com/setup_lts.x | sudo -E bash -
$ sudo apt install nodejs
$ node -v
v20.12.2
$ npm -v
v10.5.0

// install xs-dev
$ sudo apt install ssh-askpass
$ sudo npm -g i xs-dev

// install moddable SDK and ESP-IDF
$ xs-dev setup    (stops with error)
$ xs-dev setup --device esp32 (stops with error)

```

mshioji, 2024-05-07:
Hi @HipsterBrown ,

- Here is the xs-dev info: 
mshioji@raspberrypi:~ $ xs-dev dr
xs-dev environment info:
  CLI Version                0.30.4                                                                    
  OS                         Linux                                                                     
  Arch                       arm64                                                                     
  Shell                      /bin/bash                                                                 
  NodeJS Version             v20.12.2 (/usr/bin/node)                                                  
  Python Version             3.11.2 (/home/mshioji/.espressif/python_env/idf5.1_py3.11_env/bin/python) 
  Moddable SDK Version       4.6.0 (/home/mshioji/.local/share/moddable)                               
  Supported target devices   lin, esp32                                                                
  ESP32 IDF Directory        /home/mshioji/.local/share/esp32/esp-idf        

- I tried setup again, and it seems setup moddable SDK works fine with first try. ( !! )
- however, setup -- device esp32 failed with same place (/gluegun/build/index.js:15) for first, and second try.
   I do nothing special but just third try, setup esp32 succeeded....  Ummm?

- then, strange message are always happen when opening new terminal window.
  I think probably it is generated when export.sh is executed. (?)

"Not using an unsupported version of tool openocd-esp32 found in PATH: 0.12.0. To use it, run '/usr/bin/python3 /home/mshioji/.local/share/esp32/esp-idf/tools/idf_tools.py export --prefer-system' "

Thank you,

mshioji, 2024-05-11:
Hello ,

I further set up 5 Raspberry Pi's for the classroom this week, and it was no problems.
(Last week it didn't work 100% of the time, but...)

I'm not sure but maybe the Raspberry Pi OS has been updated and the problem has been resolved.(?)
Anyway, it seems it is no problem for now, so let me close this issue with this.

Thank you very much for your investigation.

closed 2024-05-11 by mshioji