HipsterBrown/xs-dev #179: Rebuild the build outputs during the update.

stc1988, 2024-09-29:
Recently, I used `xs-dev update` to update Moddable, but I noticed that only the repository is updated to the latest release, and [the build outputs are not rebuilt](https://github.com/HipsterBrown/xs-dev/blob/f15a57256ae8c1a08b523a68c759e689d90ec9ae/src/toolbox/setup/mac.ts#L143-L147). 
Rebuilding is not required with every release, but as documented in [Getting Started guide](https://github.com/Moddable-OpenSource/moddable/blob/cb204f7cd2412ec737b24416802772dc391680f7/documentation/Moddable%20SDK%20-%20Getting%20Started.md#mac-update), it might be more reliable to delete the build outputs and run `make` again.

HipsterBrown, 2024-09-30:
Thanks for creating an issue @stc1988. The build outputs should be rebuilt when updating the SDK to the latest commit from the `public` branch: https://github.com/HipsterBrown/xs-dev/blob/f15a57256ae8c1a08b523a68c759e689d90ec9ae/src/toolbox/update/mac.ts#L145-L169

When running `xs-dev update` without the `--target-branch` flag, it will default to using the latest GitHub release with pre-built tools for the SDK. Can you confirm if this is the case for you?

stc1988, 2024-09-30:
Sorry, I was looking at the wrong place in the source code.  
Also, when I tried the update procedure again, it uploaded as expected.  

Thank you for your trouble, I will close this.

closed 2024-09-30 by stc1988