HipsterBrown/xs-dev #188: Running the commands in the Development section does not create bin directory

dahenson, 2024-11-25:
I am not a full-time JS developer, so I may be missing something, but I was unable to install the development environment as described, and I'm wondering if the developer steps could be more clear.

## Steps taken:
- Clone the repository and `cd` to the directory
- Ran `pnpm install` (no errors or warnings related to xs-dev)
- Ran `pnpm link --global` (no errors)
- Ran `xs-dev` to check if it's installed, but no dice
- Running `pnpm install` again resolves fine, except this output:
```
Lockfile is up to date, resolution step is skipped
Already up to date
 WARN  Failed to create bin at /home/redacted/Projects/hipsterbrown/xs-dev/node_modules/.bin/xs-dev. ENOENT: no such file or directory, open '/home/redacted/Projects/hipsterbrown/xs-dev/node_modules/xs-dev/build/src/cli.js'
 WARN  Failed to create bin at /home/redacted/Projects/hipsterbrown/xs-dev/node_modules/.bin/xs-dev. ENOENT: no such file or directory, open '/home/redacted/Projects/hipsterbrown/xs-dev/node_modules/xs-dev/build/src/cli.js'
Done in 3.7s
```

## Workaround
I can currently run `pnpm xs-dev <command>` in order to continue local development.

## Environment
Operating system: Fedora 40
Node version: v23.3.0 (installed via asdf)
pnpm version: 9.14.2

HipsterBrown, 2024-11-27:
Using `pnpm xs-dev` is the recommended way of running the CLI during development because it will compile it on the fly, otherwise `pnpm compile` will output the expected `cli.js` in the build directory.

dahenson, 2024-11-27:
The "Development" section in the README led me to believe that running the following steps would give me what I needed to run `xs-dev` from the current branch: