HipsterBrown/xs-dev #204: An option to automatically ansswer to prompts.

stc1988, 2025-01-02:
We use `xs-dev` to setup Moddable in Github actions, but currently we need to answer prompt for  `Would you like to continue setting up and build the SDK locally? (Y/n)`.

Therefore, having an option to provide a default answer to the prompt would be helpful.

phoddie, 2025-01-02:
Doesn't xs-dev use the prebuilt binaries from the Moddable repository by default? If so, it should not be necessary for most developers to build the Moddable SDK tools themselves.

stc1988, 2025-01-02:
@phoddie 
Latest release [5.3.3](https://github.com/Moddable-OpenSource/moddable/releases/tag/5.3.3) doesn't include the prebuilt binaries.
Is this what was intended?

HipsterBrown, 2025-01-02:
There's probably a couple of ways to implement this. Some programs respond to environment variables like `CI=true` to make them non-interactive in those contexts, as well as accepting a flag like `--ci` or `--non-interactive` or `--yes` to automatically accept any interactive options. 

What do you think @stc1988?

phoddie, 2025-01-02:
@stc1988 – Oh, that's right. We normally only publish the binaries as part our monthly releases. Any incremental / patch releases only include the source code. That's inconsistent and clearly creates some potential problems. Moving forward, we'll plan to publish binaries for the incremental releases too. That's a bit more work, but seems like the right thing to do.

HipsterBrown, 2025-01-02:
@phoddie I think it's fine for incremental releases to be pushed without binaries as long as they're not tagged as the "latest" release, which is what xs-dev looks for by default. 

Also, maybe we can work together on automating those release builds based on my previous work before the binaries were included in releases: https://github.com/HipsterBrown/moddable-prebuild

phoddie, 2025-01-02:
>  I think it's fine for incremental releases to be pushed without binaries as long as they're not tagged as the "latest" release, which is what xs-dev looks for by default.

@HipsterBrown – very true. Setting the latest release is what matters. But... the usual reason we post an incremental release is because there was something so horrible that we couldn't wait. So, that needs to be set at the latest.

> Also, maybe we can work together on automating those release builds based on my previous work before the binaries were included in releases: https://github.com/HipsterBrown/moddable-prebuild

Good point. That would simplify things considerably for us. We currently provide binaries for 6 targets: Linux x86, Linux ARM, Mac 64 x86, Mac 64 ARM, Mac 64 Universal, Win 64 x86. There will be some work to get the automation to parity.

HipsterBrown, 2025-01-03:
@stc1988 Any thoughts on the implementation options I listed previously? https://github.com/HipsterBrown/xs-dev/issues/204#issuecomment-2568106547

stc1988, 2025-01-03:
@HipsterBrown 

I don’t have any particular preferences, but using
`CI=true`might eliminate the need to modify the GitHub Actions workflow.

closed 2025-01-05 by HipsterBrown

HipsterBrown, 2025-01-05:
Oh I see the error in my logic 🤦‍♂️ I have the CI setting interactive as true rather than false. Will push a fix today.

HipsterBrown, 2025-01-05:
@stc1988 This has been resolved in the release of v0.36.5.

stc1988, 2025-01-06:
It works as expected in GitHub Actions. Thank you.

PR #206 (2025-01-05): feat(setup/update): add interactive mode
Fixes #204 

Preview of new help text:

```console
USAGE
  xs-dev setup [--device esp|esp8266|darwin|mac|windows_nt|windows|win|linux|lin|esp32|wasm|pico|nrf52] [--list-devices] [--tool ejectfix] [--branch value] [--release value] [--source-repo value] [--interactive]
  xs-dev setup --help

Download and build Moddable tooling for various platform targets

FLAGS
     [--device]                         Target device or platform SDK to set up; defaults to Moddable SDK for current OS; use --list-devices for interactive selection [esp|esp8266|darwin|mac|wind [... cut]