HipsterBrown/xs-dev #209: Support ESP-IDF version required by Moddable 5.5.0

achou11, 2025-03-07:
ESP-IDF v5.4.0 is now required by the Moddable SDK (see https://github.com/Moddable-OpenSource/moddable/releases/tag/5.5.0).

I'm assuming the simple fix would be similar to #198 but would think that a more considerate approach is needed in order for xs-dev to work with older versions of the SDK.

phoddie, 2025-03-07:
xs-dev allows installing any tag of the Moddable SDK. Would this address your desire for xs-dev to support older versions the Moddable SDK?

```
xs-dev setup --release 5.3.0
```

achou11, 2025-03-07:
> xs-dev allows installing any tag of the Moddable SDK. Would this address your desire for xs-dev to support older versions the Moddable SDK?

Ah I wasn't aware of that! That probably would suffice (I don't have the personal need to stay on an older version right now)

achou11, 2025-03-07:
Does xs-dev account for that when deciding what ESP-IDF version to install though? Based on my brief explorations, seems like the answer is no, so it feels like there would ideally be some additional work to account for that `--release` properly

phoddie, 2025-03-07:
Cool. FWIW – Nick's [docs](https://xs-dev.js.org/features/setup//) are pretty solid. I scan them from time to time just to remind myself of the goodness lurking in xs-dev.

achou11, 2025-03-07:
Read the relevant docs about that `--release` tag and ran through it to figure out the answer to my question:

```sh
xs-dev setup --release 5.5.0 --device esp32
```

Attempting to run the example results in a build error (or maybe warning? can't really tell):

```sh
xs-dev run --example helloworld --device=esp32
```

```console
*** Update required to ESP-IDF v5.4 (found v5.3.1)
  See update instructions at: https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/devices/esp32.md
Expected ESP IDF v5.4, found v5.3.1
```

So I think this issue is still relevant, but agree that the `--release` tag is a nifty feature!

phoddie, 2025-03-07:
The message seems correct. Moddable SDK 5.5.0 depends on ESP-IDF v5.4, as [noted](https://github.com/Moddable-OpenSource/moddable/releases/tag/5.5.0#esp-idf) in the release notes for Moddable SDK 5.5.0.

Each version of the Moddable SDK depends on a particular version of the ESP-IDF. That's why the build checks. Espressif makes too many breaking changes for us to support a several of those in the Moddable SDK.

achou11, 2025-03-07:
> The message seems correct. Moddable SDK 5.5.0 depends on ESP-IDF v5.4, as [noted](https://github.com/Moddable-OpenSource/moddable/releases/tag/5.5.0#esp-idf) in the release notes for Moddable SDK [...]

Yes, this is understood. My point is that xs-dev doesn't handle this properly right now. It's currently hardcoded to install ESP-IDF v5.3.1 regardless of the Moddable SDK version that's being used.

phoddie, 2025-03-07:
I see. Yes, Nick and I have discussed having xs-dev refer to a file in the Moddable SDK to know which version of the ESP-IDF (or other dependency) is required. That would solve this problem and eliminate the need to update xs-dev whenever a dependency changes. I'm onboard to maintain a file (or files) for that purpose in the Moddable SDK once we work out what's needed.

HipsterBrown, 2025-03-07:
I appreciate the conversation here. The quickest way to add support for the latest Moddable release would be updating those files you've referenced @achou11, then we can start planning what type of tooling dependency file format to communicate between the Moddable SDK and xs-dev. I've done some earlier experimentation for this format, inspired by PlatformIO, here: https://github.com/HipsterBrown/xs-dev/pull/120 , which goes much further into making xs-dev a generic task runner for embedded tooling setups. 

I just noticed there's a build variable used in mcconfig to set the `EXPECTED_ESP_IDF` version: https://github.com/Moddable-OpenSource/moddable/blob/b11c527e1e2fac50deea2e375ddaf5995f670e9c/tools/mcconfig/make.esp32.mk#L41
Given how long it's been in the Moddable SDK codebase, it seems like a reliable source of truth that xs-dev could parse from that make file. What do you think @phoddie ?

@achou11 Would you be open to creating a PR with those initial changes to support the latest Moddable release?

achou11, 2025-03-07:
> Would you be open to creating a PR with those initial changes to support the latest Moddable release?

Done! See #210

closed 2025-03-07 by HipsterBrown

phoddie, 2025-03-07:
> Given how long it's been in the Moddable SDK codebase, it seems like a reliable source of truth that xs-dev could parse from that make file. What do you think @phoddie ?

You could. But, I'd rather not be on the hook for maintaining that. One well-intentioned change would break xs-dev. It would be trivial to create $MODDABLE/build/versions.json and start with:

```json
{
	"external": {
		"esp-idf-tag": "v5.4.0"
	}
}
```

PR #210 (2025-03-07): feat(esp32): update esp-idf version to 5.4.0
Closes #209