HipsterBrown/xs-dev #212: xs-dev run fails with 'MAKE_FRAGMENT not found' error on Ubuntu

mshioji, 2025-03-26:
Hello, first of all, I really appreciate your work on this great tool. It has been very helpful for my projects.

Recently, I found that the `xs-dev setup` command doesn't seem to work properly on Ubuntu 24.04 and Ubuntu 22.04.  
I also tried it on a Raspberry Pi, and it works fine with both the latest Bookworm and the previous Bullseye versions.

### 1. Dependencies  
It seems that I need to install `python3-venv` and `curl` before executing `xs-dev setup` on a newly installed Ubuntu system.
It might be better if xs-dev setup could automatically install these dependencies too, or at least check for them and provide a warning.

### 2. Environment Setup  
After setting up the Moddable environment with `$ xs-dev setup`, I tried running `$ xs-dev run --example helloworld`,  
but I encountered the following error:

```
$ xs-dev run --example helloworld
⠹ Building and deploying project /home/mshioji/.local/share/moddable/examples/helloworld on linux
### Error: MAKE_FRAGMENT not found: unknown platform "linux"!
Error: Command failed with exit code 1: mcconfig -m -p linux -t all -o /home/mshioji/.local/share/moddable/build -d
    at makeError (/usr/lib/node_modules/xs-dev/node_modules/execa/lib/error.js:60:11)
    at handlePromise (/usr/lib/node_modules/xs-dev/node_modules/execa/index.js:118:26)
    at process.processTicksAndRejections (node:internal/process/task_queues:105:5) {
  shortMessage: 'Command failed with exit code 1: mcconfig -m -p linux -t all -o /home/mshioji/.local/share/moddable/build -d',
  command: 'mcconfig -m -p linux -t all -o /home/mshioji/.local/share/moddable/build -d',
  escapedCommand: 'mcconfig -m -p linux -t all -o "/home/mshioji/.local/share/moddable/build" -d',
  exitCode: 1,
  signal: undefined,
  signalDescription: undefined,
  stdout: undefined,
  stderr: undefined,
  failed: true,
  timedOut: false,
  isCanceled: false,
  killed: false
```

In this case, `xs-dev doctor` shows the following:

```
$ xs-dev dr
xs-dev environment info:
  CLI Version                0.37.0                                      
  OS                         Linux                                       
  Arch                       x64                                         
  Shell                      /bin/bash                                   
  NodeJS Version             v22.14.0 (/usr/bin/node)                    
  Python Version             3.12.3 (/usr/bin/python)                    
  Moddable SDK Version       5.5.0 (/home/mshioji/.local/share/moddable) 
  Supported target devices   linux                                
  ESP32 IDF Directory        /home/mshioji/.local/share/esp32/esp-idf    

If this is related to an error when using the CLI, please create an issue at "https://github.com/hipsterbrown/xs-dev/issues/new" with the above info.
```

I suspect that this issue might be related to incorrect environment variable handling or some misconfiguration.  
In this condition, the Moddable original `mcconfig` command is also missing.  

when I manually add the environment variables from `~/.local/share/xs-dev-export.sh` to `~/.bashrc`,  
`mcconfig` starts working, but `xs-dev run` still fails.

I'm wondering what the difference is between Ubuntu and Raspberry Pi, since the command works fine on Raspberry Pi.

Thank you for your help.

mshioji, 2025-03-27:
Update:
I originally thought this issue was specific to Ubuntu, but I just tested xs-dev run on Raspberry Pi and encountered the similar error.

xs-dev setup completes successfully on Raspberry Pi.
However, running xs-dev run --example helloworld results in the following error.
Moddable mcconfig also I got "bash: mcconfig: command not found."

```
$ xs-dev run --example helloworld
⠦ Building and deploying project /home/riot/.local/share/moddable/examples/hello⠧ Building and deploying project /home/riot/.local/share/moddable/examples/helloworld on linux
Error: Command failed with exit code 1: mcconfig -m -p linux -t all -o /home/riot/.local/share/moddable/build -d
    at makeError (/usr/lib/node_modules/xs-dev/node_modules/execa/lib/error.js:60:11)
    at handlePromise (/usr/lib/node_modules/xs-dev/node_modules/execa/index.js:118:26)
    at process.processTicksAndRejections (node:internal/process/task_queues:105:5) {
  shortMessage: 'Command failed with exit code 1: mcconfig -m -p linux -t all -o /home/riot/.local/share/moddable/build -d',
  command: 'mcconfig -m -p linux -t all -o /home/riot/.local/share/moddable/build -d',
  escapedCommand: 'mcconfig -m -p linux -t all -o "/home/riot/.local/share/moddable/build" -d',
  exitCode: 1,
  signal: undefined,
  signalDescription: undefined,
  stdout: undefined,
  stderr: undefined,
  failed: true,
  timedOut: false,
  isCanceled: false,

```
So this issue might not be limited to Ubuntu but rather affect Linux systems in general.
Sorry for the confusion, and thank you for your support!

HipsterBrown, 2025-03-27:
Thank you for the update and all the information about your setup. 

Can you try sourcing the `xs-dev-export.sh` file in your terminal and see if that fixes the issue? 

We recently stopped automatically including that export file in the user's shell profile and sourcing it in the memory of xs-dev before executing commands to reduce some complexity. I'll work on recreating this setup on my Ubuntu laptop to debug. 

You're correct that xs-dev should ensure curl and python3-venv are available on Linux during the setup process. 👍

mshioji, 2025-03-28:
First of all, thank you for looking into this issue! I really appreciate your support.

Here’s the situation:

1. Initially, running xs-dev run --example helloworld fails with the error:
"MAKE_FRAGMENT not found: unknown platform 'linux'!"

2. After sourcing ~/.local/share/xs-dev-export.sh, mcconfig starts working fine.
However, even after sourcing, xs-dev run --example helloworld still doesn't work. 

3. It looks like the Moddable environment itself was set up successfully with xs-dev setup.
In fact, I am using this Moddable environment with Node-RED MCU, and it runs without issues.
(Because of this, I initially thought everything was working fine on my Raspberry Pi...)

I’ll attach the full debug logs below. Let me know if there's anything else I should check!

Thanks!

- xs-dev dr
```
mshioji@chubbycat:~$ xs-dev dr
xs-dev environment info:
  CLI Version                0.37.0                                      
  OS                         Linux                                       
  Arch                       x64                                         
  Shell                      /bin/bash                                   
  NodeJS Version             v22.14.0 (/usr/bin/node)                    
  Python Version             3.12.3 (/usr/bin/python)                    
  Moddable SDK Version       5.5.0 (/home/mshioji/.local/share/moddable) 
  Supported target devices   linux                                       

If this is related to an error when using the CLI, please create an issue at "https://github.com/hipsterbrown/xs-dev/issues/new" with the above info.
```
- before sourcing: xs-dev
```
mshioji@chubbycat:~$ xs-dev run --example helloworld
⠙ Building and deploying project /home/mshioji/.local/share/moddable/examples/helloworld on linux
### Error: MAKE_FRAGMENT not found: unknown platform "linux"!
Error: Command failed with exit code 1: mcconfig -m -p linux -t all -o /home/mshioji/.local/share/moddable/build -d
    at makeError (/usr/lib/node_modules/xs-dev/node_modules/execa/lib/error.js:60:11)
    at handlePromise (/usr/lib/node_modules/xs-dev/node_modules/execa/index.js:118:26)
    at process.processTicksAndRejections (node:internal/process/task_queues:105:5) {
  shortMessage: 'Command failed with exit code 1: mcconfig -m -p linux -t all -o /home/mshioji/.local/share/moddable/build -d',
  command: 'mcconfig -m -p linux -t all -o /home/mshioji/.local/share/moddable/build -d',
  escapedCommand: 'mcconfig -m -p linux -t all -o "/home/mshioji/.local/share/moddable/build" -d',
  exitCode: 1,
  signal: undefined,
  signalDescription: undefined,
  stdout: undefined,
  stderr: undefined,
  failed: true,
  timedOut: false,
  isCanceled: false,
  killed: false
```
- before sourcing : mcconfig
```
mshioji@chubbycat:~$ cd ~/.local/share/moddable/examples/helloworld/
mshioji@chubbycat:~/.local/share/moddable/examples/helloworld$ mcconfig -d -m -p lin
Command 'mcconfig' not found, did you mean:
  command 'cconfig' from deb xrootd-server (5.6.4-1)
  command 'mconfig' from deb mono-devel (6.8.0.105+dfsg-3.5ubuntu1)
  command 'ccconfig' from deb libconvert-binary-c-perl (0.84-1build2)
Try: sudo apt install <deb name>
```
- sourcing xs-dev-export.sh
```
mshioji@chubbycat:~/.local/share/moddable/examples/helloworld$ cd ~
mshioji@chubbycat:~$ source ~/.local/share/xs-dev-export.sh
```
- xs-dev
```
mshioji@chubbycat:~$ xs-dev run --example helloworld
⠙ Building and deploying project /home/mshioji/.local/share/moddable/examples/helloworld on linux
### Error: MAKE_FRAGMENT not found: unknown platform "linux"!
Error: Command failed with exit code 1: mcconfig -m -p linux -t all -o /home/mshioji/.local/share/moddable/build -d
    at makeError (/usr/lib/node_modules/xs-dev/node_modules/execa/lib/error.js:60:11)
    at handlePromise (/usr/lib/node_modules/xs-dev/node_modules/execa/index.js:118:26)
    at process.processTicksAndRejections (node:internal/process/task_queues:105:5) {
  shortMessage: 'Command failed with exit code 1: mcconfig -m -p linux -t all -o /home/mshioji/.local/share/moddable/build -d',
  command: 'mcconfig -m -p linux -t all -o /home/mshioji/.local/share/moddable/build -d',
  escapedCommand: 'mcconfig -m -p linux -t all -o "/home/mshioji/.local/share/moddable/build" -d',
  exitCode: 1,
  signal: undefined,
  signalDescription: undefined,
  stdout: undefined,
  stderr: undefined,
  failed: true,
  timedOut: false,
  isCanceled: false,
  killed: false
```
- mcconfig 
```
mshioji@chubbycat:~$ cd ~/.local/share/moddable/examples/helloworld/
mshioji@chubbycat:~/.local/share/moddable/examples/helloworld$ mcconfig -d -m -p lin
# cc lin_xs.c
# cc xsAll.c
# cc xsAPI.c
# cc xsArguments.c
# cc xsArray.c
# cc xsAtomics.c
# cc xsBigInt.c
# cc xsBoolean.c
# cc xsCode.c
# cc xsCommon.c
# cc xsDataView.c
# cc xsDate.c
# cc xsDebug.c
# cc xsError.c
# cc xsFunction.c
# cc xsGenerator.c
# cc xsGlobal.c
# cc xsJSON.c
...
```

HipsterBrown, 2025-03-28:
Thank you for the detailed steps. I believe I know the cause of the issue. Some recent refactoring led to `linux` being used as the platform type instead of `lin` as expected by mcconfig when running or building programs. To resolve this, the `build.ts` file needs to be updated to map the OS platform target name expected by mcconfig:

- mac == mac
- linux == lin
- windows == win

That logic would probably go around this line: https://github.com/HipsterBrown/xs-dev/blob/main/src/toolbox/build/index.ts#L108

dahenson, 2025-07-16:
The dependency issues as well as the setup script issues are being worked on in #195 

These are all things I encountered when attempting to install on Fedora. Once those are merged, I will work to make sure the rest of the commands and stack work well with as many Linux distributions as possible to help with adoption among other developers.