HipsterBrown/xs-dev #215: Track the dependent version of ESP-IDF

stc1988, 2025-08-16:
In #214 , I upgraded the ESP-IDF version to match the latest Moddable, but having to handle it every time it’s updated feels cumbersome. Would it be possible to read the ESP-IDF version used from the Moddable repository and automatically track it?  
cc: @phoddie

phoddie, 2025-08-16:
Good idea. We've discussed this before but never quite arrived at a solution. I'm open to adding that information to the Moddable SDK in an easily parsable file. It would be great if we could do that in a way that also allowed it to be used by our make files (see [make.esp32.mk](https://github.com/Moddable-OpenSource/moddable/blob/af9b5ba9e1a97fddb25f78d0e2ebd6fb70bb6eb2/tools/mcconfig/make.esp32.mk#L42C1-L42C17), for example) so that there's just one file to update when migrating to a new version.

stc1988, 2025-08-17:
I can think of two ways.

1. It is a simple key-value format that lists such software components and versions.

``` 
ESP-IDF=5.5
```

2. By using the [SPDX JSON format](https://github.com/spdx/spdx-spec/blob/develop/examples/jsonld/package_sbom.json), we can manage not only version information but also other details.
However, since I am not very familiar with SPDX, I cannot think of other ways to make use of it.

closed 2025-08-22 by HipsterBrown